prusa3d/PrusaSlicer #1614: Layers made out of bridging and overhang perimeters out of nowhere (1.42a2)

glowtape, 2019-01-06:
### Version
1.42 alpha 2

### Operating system type + version
Windows 10 Pro Build 17763.195

### Behavior
Slicing a handle, whose insides are hollowed with large slots, the application seems to have problems dealing with the model for unknown reasons.

First at layer at layer 22-24 it creates top infill, then at layer 25, it suddenly uses bridging to fill said slots (layer 25 is kind of a floating layer), followed by layer 26 being all bridging and overhang perimeters, to then just continue slicing the model as normal but starting with solid infill.

The model slices just fine in 1.41. The model was also inspected in Meshmixer, it isn't two part, nor are there breaks in the polygons at the height slic3r stumbles.

It's the first time this appears. I've sliced and printed other models with 1.42a2 without issues of that sort so far. It's reproducible here by starting slic3r clean, just dropping the handle model on the plate and immediately slicing, without any other operations in the app.

[image]

[image]

#### STL/Config (.ZIP) where problem occurs
[slbug.zip](https://github.com/prusa3d/Slic3r/files/2730994/slbug.zip)

glowtape, 2019-01-11:
Still does it as of Alpha 3.

glowtape, 2019-01-12:
Running the model manually through Netfabb makes it work.

That said, in 1.41 it slices fine without preprocessing.

bubnikv, 2019-01-14:
Fixed with cf8e4fd7b0c8374c98daffa59e46ae2d5c14651a
Closing.

closed 2019-01-14 by bubnikv