prusa3d/PrusaSlicer #1620: Modifier Mesh does not work correctly with Elephant Foot correction

fdurrer, 2019-01-07:
### Version
Slic3rPE 1.42.0 alpha2

### Operating system type + version
Win 7

### Behavior
The elephant foot correction handles the modifier mesh as separate part. This leads to a gap of 2 times the correction between the unmodified part and the part with modifier mesh. See attached pictures.

Part with modifier mesh
[image]

Sliced first layer without elephant foot correction
[image]

Sliced first layer with elephant foot correction
[image]

bubnikv, 2019-03-05:
There was a bug in alpha2, which allowed you to specify the object level parameters for a modifier. This is now fixed in alpha7, so you are no more able to set the elephant foot compensation at the volume or modifier level.

Closing.

fdurrer, 2019-03-05:
Hi bubnikv

Thanks for feedback.
Unfortunately, the issue persists in 1.42.0 alpha7 win64:
[image]

Elephant foot correction of 0.2mm:
[image]

Inside looks good (modifier should only affect infill density):
[image]

But also top solid infill and solid infill looks like the modifier is handeld as separate part
[image] 
[image]

My settings
[image]

I can also upload the 3mf file if this helps.

bubnikv, 2019-03-06:
Uploading a 3MF helps each and every time.

Your issue is in a way similar to https://github.com/prusa3d/Slic3r/issues/1143

I am looking into the elephant foot compensation & modifier meshes right now. It is a similar issue to applying the elephant foot compensation to a multi-material print.

bubnikv, 2019-03-06:
The elephant foot over modifier meshes or multi-material should be fixed with 5e582efc5ce3fcbbe92eb49771f94ba669730551

It will be part of the 1.42.0-beta.

fdurrer, 2019-03-07:
Sorry for the delayed reply

Attached the 3mf file to reproduce.
[xyzCalibration_cube.zip](https://github.com/prusa3d/Slic3r/files/2943301/xyzCalibration_cube.zip)

I agree, that the issue is similar to #1143
I get the same when I add perimeter settings to the modifier
[image]

fdurrer, 2019-03-07:
Update:
Seems to be a very old bug. I made some tests with several versions back to 1.37.2-prusa3d-win64 and there I get a similar behavior:
[image]
(only infill chanched in modifier)

[image]
(numbers of perimeters chanched in modifier)

jhoblitt, 2019-03-25:
To be clear, the "break" in perimeters are the edge of a modifier volume isn't related to elephant foot compensation. I discovered this behavior the other day only after a part finished printing.  Perimeter modifiers are virtually unusable because of this.

I'm guessing (???) that a modifier adding additional perimeters over the print default will be easier to handle as most users would expect as regular perimeters can be plotted before prior to any additional lines inside a modifier volume.

bubnikv, 2019-11-01:
Splitting of perimeters and infill at the top / bottom surfaces was fixed in PrusaSlicer 2.1.0. Closing.

closed 2019-11-01 by bubnikv