prusa3d/PrusaSlicer #1621: Estimate elapsed time of slice.

badger6, 2019-01-08:
have option to show total elapsed time at each slice to facilitate knowing when a filament color change will take place.  I would like to be able to leave the printer and return when it is necessary to change the filament for a color change.

_Is this a new feature request?_ yes

EXOgreen, 2019-01-08:
@badger6 with the latest Slic3r 1.42.0-alpha# you will be able to add color change layers that pauses the print before it starts the next layer.  I suggest reading [this](https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-alpha1) for all the changes, they have a [video](https://youtu.be/qoAL6zwERg8) linked too, the timestamp for color changing is [6:56](https://youtu.be/qoAL6zwERg8?t=416)

Duplicate of #1533

jmd73, 2019-01-08:
@EXOgreen I think he knows that. What he want is an option to show remaining printtime until next colorchange, instead of showing printtime remaining för the whole print. För example ön a 5 hour print with multiple color changes you can see that you have 30 minutes until next colorchange (so you can dö other stuff during that time.) After first color change you get the time to the next, and so on. I like that idea.

EXOgreen, 2019-01-08:
That actually is a great idea, thanks for clarifying!

rcmaniac25, 2019-05-07:
I too would like this feature. I have a print that I'd like to start and come back at a specific time to do a color change. Knowing when I need to do that would be great (I usually wear headphones, so simply pausing and beeping won't grab my attention)

JangusKhan, 2019-07-29:
I use color change a lot. It's very well implemented in the current (Prusaslicer 2.0.0) release. It would be fantastic if I could see the time estimate for when each color swap will happen after tweaking settings and before exporting gcode. In this way, I can see whether or not I have enough time to start a color swapped print before going to bed or what have you. Even better: Alternate print time remaining and time-till-swap on the LCD.

bubnikv, 2019-07-29:
Time of color change will be displayed by the upcoming PrusaSlicer 2.1 

> Alternate print time remaining and time-till-swap on the LCD.

We have discussed this feature with the firmware team. It is useful indeed and it should not be too difficult to implement.

rtyr, 2020-03-30:
Implemented. Closing.

closed 2020-03-30 by rtyr

bendavis78, 2022-06-11:
Where was this implemented? I googled "prusa slicer estimated time to color change" and this issue was the first hit. I'm having trouble finding the time estimate in prusa slicer (it only shows the total print time, not time to layer)

badger6, 2022-06-11:
I don’t remember when it was implemented, but I’m using version 2.4.0 and it is implemented.  The color change schedule appears in the “color print” box in the upper left corner of the viewing area in the slicer “preview” screen. The box will appear as soon as you specify a color change by clicking on the “+” hexagon adjacent to the slider on the right side of the “preview” area.

rcmaniac25, 2022-06-12:
Yes, when you go to the slice preview screen and do a filament swap, you get a preview that tells you how long each filament swap will take and how much total time is remaining. In addition, new gcode gets exported and on the latest firmware (at least MK3*), it will actually show you the total time remaining AND how long until the filament swap happens. So you not only get a preview, but can literally just look at the printer and go "ah, I still have 2h. I'll come back later"