prusa3d/PrusaSlicer #1622: Error while translating strings and more.

AndylgTom, 2019-01-08:
**Version**
Slic3rPE-1.42.0-alpha2

**Operating system type + version**
Operating system WIN10 64bit

**Behavior**
1. If you translate the lines box - slab, when you select them in the program a window pops up with an error.
[image]
[image]

2. Incorrect encoding in these windows \ tabs. (Cyrillic)

[image]
[image]

3. The "Mashine limits" tab is not translated, although the translation of these lines is present in the translation file (In any language). 

[image]

YuSanka, 2019-01-08:
Hello @AndylgTom, 
Thanks for your report. 

For the questions #1 and #3 - the translation hasn't been finished at the moment and it's up to our contributors.  

For the question #2 - there are two separate bugs: 
       a) Cyrillic-encoded file names inside object lists is already fixed, but 
       b) ImGui localization - is a bigger issue and it will take longer for us to fix.

bubnikv, 2019-03-02:
In Slic3r 1.42.0-alpha7, the ImGUI fonts are pre-rendered into a texture based on the user language. For Russian and Ukrainian, the ImGUI renders 7bit ascii and Cyrillic. 

Closing.

closed 2019-03-02 by bubnikv