prusa3d/PrusaSlicer #1624: Slicing for MMU Multi vs MMU Single misses features

wcraigtrader, 2019-01-08:
### Version

Slic3rPE-1.42.0-alpha2+linux64-full-201901041550.AppImage

### Operating system type + version

Linux, Ubuntu 16.04.5 LTS

### Behavior

When slicing 3DBenchy for the Prusa Mk3, the generated GCode makes an attempt at rendering the name plate. When slicing for the Prusa MK3 + MMU, the generated GCode drops the name plate entirely, revealing the bottom of the fishing rod holder through the back wall of the boat.

### How to Reproduce

1. Start Slic3r-PE fresh.
1. Select the `Original Prusa i3 MK3` printer profile.
1. Select the `0.15mm OPTIMAL MK3` print settings.
1. Import the attached AMF file [3DBenchy-multi.zip.amf](https://github.com/prusa3d/Slic3r/files/2737794/3DBenchy-multi.zip.amf.zip).
1. Zoom in on the name plate at the rear of the model (`#3DBenchy`).
1. Switch to the Preview. You'll see that the name plate is partially rendered, which is expected because the MK3 (and the MK2) just aren't precise enough to render the name plate with a 0.4 nozzle.

[image]

1. Swith to the Editor.
1. Select the `Original Prusa i3 MK3 MMU2` printer profile.
1. Zoom back in on the name plate.
1. Switch to the Preview. You'll see that the name plate has vanished entirely, and that there's a V-shaped hole in the stern of the boat, and that you can see the bottom of the fishing rod holder.

[image]

#### STL/Config (.ZIP) where problem occurs

[config.ini](https://github.com/prusa3d/Slic3r/files/2737869/config.ini.txt)
[3DBenchy-multi.zip.amf](https://github.com/prusa3d/Slic3r/files/2737794/3DBenchy-multi.zip.amf.zip)

bubnikv, 2019-01-10:
What do you expect? Why do you think it is an error?

You see the blue tube is sticking out of the boat hull.

bubnikv, 2019-01-10:
By the way, is the label there at the multi-material hull of the boat?

wcraigtrader, 2019-01-10:
I expected to see a partially rendered name plate, just like when slicing with one filament.
I think it's an error because the name plate doesn't appear as part of the sliced results, thereby allowing the interior details of the fishing rod holder to appear.

bubnikv, 2019-01-29:
The label cannot be extruded with its own extruder, because it is very thin, thinner than a perimeter.
If you assign it an extruder 1 (the extruder that you assigned to the hull), then they will be merged and at least something will show up at the hull.

bubnikv, 2019-03-02:
The object is thinner than a perimeter, therefore it is not printed. The software behaves correctly. Closing.

closed 2019-03-02 by bubnikv

marcantoinebeaulieu, 2020-11-22:
> The label cannot be extruded with its own extruder, because it is very thin, thinner than a perimeter.
> If you assign it an extruder 1 (the extruder that you assigned to the hull), then they will be merged and at least something will show up at the hull.

That does not work. Changing any single colour, or using any other tool, say on the chimney only, will cause the nameplate to disappear. This still happens on 2.3.0-alpha4.