prusa3d/PrusaSlicer #1626: Slic3r 1.42.0-alpha2 - is missing extrusion multiplier option

EyalSi, 2019-01-09:
Does anyone know whether configuring extrusion multiplier is possible in this version? in previous versions it was under Filament Settings.

FidelCapo, 2019-01-09:
The extrusion multiplier setting is still located in the Filament settings, but it requires a slicer running in Advance mode to display it. The mode can be selected in Configuration Menu - Mode - Simple/Advanced/Expert. Did it solve your issue?

aligator, 2019-01-09:
Maybe there should be a message that there are different modes now in the wizard on first run. I think that there are more people who don't know there are different modes now.

bubnikv, 2019-01-10:
We are working on a UI for switching the mode, persisting on the plater.
Closing.

closed 2019-01-10 by bubnikv