prusa3d/PrusaSlicer #1627: XYZ values doesn't change after object manipulation

tiboine, 2019-01-09:
### Version
Version 1.42.0-alpha2+win64

### Operating system type + version
Win 10 64 build 17763.253

### Behavior

Model height (Z) is based on the original model orientation. So when I rotate the model 90 degrees, the Z height doesn't update to its new Z orientation. 

[image]
[image]

bubnikv, 2019-01-10:
@FidelCapo is this a new issue? Has it been fixed in our development builds?

bubnikv, 2019-01-10:
The position/rotation/size factors that you are entering into the table are applied to the loaded model in the following order:

1) Scale
2) Rotation
3) Translation

That means, the scale is applied in the coordinate system of the object before it is rotated. If you rotate the object in X by 90 degrees, the Z axis of the object used for scaling rotates with the object. This is by design.

We may discuss whether it is a good idea or not, it is one of the two ways to skin the cat. The other way is to scale in the world coordinate system, but then an anisotropic scaling (for example by the world X axis only) would introduce skew along some diagonal of the original object if applied on an object rotated by an angle different from multiples of 90 degrees.

To make the concept clearer, you will see a hint arrow displayed over the 3D object, when one of the position/rotation/scale input fields get focused. Unfortunately, there is a bug to be fixed for the size fields, where the hints are displayed for a wrong axis. We shall fix that. For the Scale factors field, the visual hints are displayed correctly

Does this explanation make the concept clear?

tiboine, 2019-01-10:
This isn't an issue in 1.41.2+win64.

EXOgreen, 2019-01-13:
I think it is quite confusing/annoying that if I want to scale an item that has been rotated using place face to half its height, it's almost phisically impossible to do so. Please change it back so that we can move/rotate/scale based o  the wold axis not the item axis, or make that an option somewhere

bubnikv, 2019-01-29:
We will likely implement scaling in world coordinates in the next release, not in this one.

bubnikv, 2019-05-14:
Rotation and scaling of object instances in the World coordinate system is supported by PrusaSlicer 2.0.0-rc. Closing.

closed 2019-05-14 by bubnikv