prusa3d/PrusaSlicer #1631: 1.42.0-dev erroneous perimeter

jurriaan, 2019-01-09:
### Version
5aec48418a0a55c0a9d5d9df4f6f4e680647fcee (build myself to get the latest fixes)

### Operating system type + version
Linux (Arch)

### Behavior
When slicing a specific STL I noticed there were perimeters in the object that shouldn't be there.

These perimeters are only on the topmost layers of one corner of the object.

This is not the case in Slic3r 1.41.2

| v1.41.2 | v1.42.0-alpha-2 |
| ------- | --------------- |
| [image] | [image] |

#### STL/Config (.ZIP) where problem occurs
[stl_and_config.zip](https://github.com/prusa3d/Slic3r/files/2742789/stl_and_config.zip)

jurriaan, 2019-01-12:
The issue is still present in the official v1.42.0-alpha3 release

[image]

bubnikv, 2019-01-14:
This is a long standing bug, where an extrusion is incorrectly drawn from the end of the last wipe move. Interestingly enough, this bug is in Slic3r at least since 1.2.9, but lucky enough  it only occurred for single perimeter, no infill prints with wipe after retract enabled, and only if the two successive slices were discretised exactly the same, which is quite unlikely.

Fixed with 54299d8eb0db7910b73aba7101484a10823dc32d
Closing

closed 2019-01-14 by bubnikv