prusa3d/PrusaSlicer #1632: macos crash on WipeTower when using multi extruder + tower + support

tedder, 2019-01-09:
### Version
1.4.2-alpha2

### Operating system type + version
macos 10.14.2

### Behavior
1. import a STL
2. enable support, enable wipe tower
3. override the extruder for support
4. set the main extruder to a nondefault
5. while slicing, it crashes.

Steps 3 and 4 can be swapped, or 2 can be done last, it still crashes.

it happens that I'm trying to use extruder 1 (with a PLA profile) for support and extruder 5 (with a PETG profile) when doing this. I don't think it's relevant, I've reproduced going between 1 (as support) and 2 (both PLA) and 4 (as support) and 5 (both PETG) and 1 (as support) and 5 (a mix of PLA/PETG) and it still crashes.

If I *DISABLE* the wipe tower it works.

I'm getting really tired of reproducing this crash now :)

I thought it might be related to the OpenGL stuff, but same behavior on 'legacy' and not.

# crash log
```
Thread 6 Crashed:
0   com.prusa3d.slic3r/           	0x00000001099df377 Slic3r::WipeTowerIntegration::append_tcr(Slic3r::GCode&, Slic3r::WipeTower::ToolChangeResult const&, int) const + 103
1   com.prusa3d.slic3r/           	0x00000001099e162e Slic3r::WipeTowerIntegration::tool_change(Slic3r::GCode&, int, bool) + 158
2   com.prusa3d.slic3r/           	0x00000001099ecad8 Slic3r::GCode::process_layer(__sFILE*, Slic3r::Print const&, std::__1::vector<Slic3r::GCode::LayerToPrint, std::__1::allocator<Slic3r::GCode::LayerToPrint> > const&, Slic3r::LayerTools const&, unsigned long) + 6264
3   com.prusa3d.slic3r/           	0x00000001099e82ec Slic3r::GCode::_do_export(Slic3r::Print&, __sFILE*) + 17644
4   com.prusa3d.slic3r/           	0x00000001099e2a74 Slic3r::GCode::do_export(Slic3r::Print*, char const*, Slic3r::GCodePreviewData*) + 644
5   com.prusa3d.slic3r/           	0x0000000109af67c5 Slic3r::Print::export_gcode(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, Slic3r::GCodePreviewData*) + 197
6   com.prusa3d.slic3r/           	0x0000000109c66e2a Slic3r::BackgroundSlicingProcess::process_fff() + 218
7   com.prusa3d.slic3r/           	0x0000000109c67e0c Slic3r::BackgroundSlicingProcess::thread_proc() + 380
8   com.prusa3d.slic3r/           	0x0000000109c68f20 void* std::__1::__thread_proxy<std::__1::tuple<Slic3r::BackgroundSlicingProcess::start()::$_1> >(void*) + 96
9   libsystem_pthread.dylib       	0x00007fff76ec7305 _pthread_body + 126
10  libsystem_pthread.dylib       	0x00007fff76eca26f _pthread_start + 70
11  libsystem_pthread.dylib       	0x00007fff76ec6415 thread_start + 13
```

#### STL/Config (.ZIP) where problem occurs
including stl and config (and the entire crash log) in [slic3r-crash-supporting-files.zip](https://github.com/prusa3d/Slic3r/files/2742930/slic3r-crash-supporting-files.zip). Also here's a screen shot of the area where it tends to crash- note 'generate support' is turned off in this case: 
[image]

tedder, 2019-01-09:
better screenshot:

[image]

bubnikv, 2019-01-10:
In 1.42 we made some illegal things too easy, so they will now hunt us :-)

[image]

The Single extruder - multiple material mode generates non-soluble supports with layers, which are not synchronized with the wipe tower. In that case the extruder IDs for the support material have to be set to zero indicating to use whatever extruder is active. For the soluble or partially soluble supports to work, you have to enable the "Synchronize with object layers" feature, then you may assign the extruders as you wish.

We shall enforce the non-soluble supports rule with a check.

bubnikv, 2019-01-10:
I suppose you received the following message when you tried to set the support extruders in the fields marked at the previous screen shot.

[image]

Sorry, but for now we shall just extend the check to the parameter overrides.

tedder, 2019-01-10:
I didn't get that warning. Not sure if it was the order of operations or what.

bubnikv, 2019-01-10:
You would get the warning, if you had set all the extruders at the Print
preset page.

I am going to add the missing checks.

bubnikv, 2019-01-10:
Fixed with 2b0e1633cf2b1ff8c7906c3c117d7dada001cbee
Now you will not get a crash, but an error message explaining, why you cannot slice with such a configuration.

If you really need such configuration, you will likely want to have the "synchronize object layers with support layers" checkbox configurable for the non-soluble support interfaces as well. If you decide that you need that, please open another issue.

Actually there are similar issues open already:
https://github.com/prusa3d/Slic3r/issues/553
https://github.com/prusa3d/Slic3r/issues/957

The technical reason to not allow synchronized support layers is, that it gives another constraint for fulfilling the support to object separation gap requirement.

closed 2019-01-10 by bubnikv