prusa3d/PrusaSlicer #1633: 1.42 A2 slicing error

screwyluie, 2019-01-10:
### Version
Slic3r 1.42.0-alpha2 (64bit)

### Operating system type + version
Windows 10

### Behavior
* _Describe the problem_

slicing error. There's a layer in the middle of the mesh that shouldn't be there. This error is not present in slic3r 1.41.2, cura, netfabb, meshmixer, or any other mesh tools.

* _Steps needed to reproduce the problem_

slice the attached file

* _Actual Results_

https://imgur.com/a/2NXdd7w

[image]
[image]

_Is this a new feature request?_

no

#### STL/Config (.ZIP) where problem occurs

[slic3rerror.zip](https://github.com/prusa3d/Slic3r/files/2744050/slic3rerror.zip)

glowtape, 2019-01-10:
Looks same as my issue #1614.

screwyluie, 2019-01-10:
indeed it does. I searched for slicing errors but didn't find that one... but they do look the same.

glowtape, 2019-01-12:
You could try going to Netfabb and process the model there and see what it does in slic3r. It made my model work in 1.42, but that's just a work-around.

https://service.netfabb.com/

bubnikv, 2019-01-14:
Fixed with cf8e4fd7b0c8374c98daffa59e46ae2d5c14651a
Closing.

closed 2019-01-14 by bubnikv