prusa3d/PrusaSlicer #1641: Gnome: can't resize secondary column?

tedder, 2019-01-12:
### Version
1.42.0-alpha3 appimage

### Operating system type + version
Ubuntu 18.10, Gnome 3.30.1

### Behavior
When I open Slic3r, the secondary column is cut off. I can't resize it. Normally I have my display zoom at 200% (because I'm on a 5k monitor); even at 100% it is too narrow.

Even the 'settings' left-hand listing is too narrow. See screenshots below.

right main column at 200%:
[image]

left 'settings' column at 200%:
[image]

right main column at 100%:
[image]

left 'settings' column at 100%:
[image]

#### STL/Config (.ZIP) where problem occurs
n/a

jdevelop, 2019-01-21:
I have noticed something pretty similar on my `eDP1 connected primary 3840x2160+0+0 (normal left inverted right x axis y axis) 340mm x 190mm`

[image]

It would be nice to have some "stretch" for wide-screen HIDPI displays.

bubnikv, 2019-03-02:
The HiDPI issues shall be fixed in the upcoming Slic3r 1.42.0-alpha7. Closing.

closed 2019-03-02 by bubnikv