prusa3d/PrusaSlicer #1642: 1.42.0-alpha3 - can only open 3mf files?

Mirarkitty, 2019-01-12:
### Version
1.42.0-alpha3

Slic3rPE-1.42.0-alpha3+linux64-full-201901111425.AppImage

### Behavior
* "open" dialogue only lets me select "*.3mf"

[image]

ddavidebor, 2019-01-12:
that's right, there is a import menu now.

[image]

thenickdude, 2019-01-17:
You can also just drag and drop STL files onto the build plate.

This menu would probably be clearer if it was renamed to "open project...".

bubnikv, 2019-03-02:
The Open menu was renamed to "Open Project". I hope it is clearer now.
Closing.

closed 2019-03-02 by bubnikv