prusa3d/PrusaSlicer #1644: 1.42.0-alpha3 - right side of user interface in plater is not right

Mirarkitty, 2019-01-12:
### Version
Slic3rPE-1.42.0-alpha3+linux64-full-201901111425.AppImage

### Behavior
Right side of user interface is not rightly positioned, input fields are not fully showing and it can't be rescaled.
[image]
[image]

EXOgreen, 2019-01-13:
Duplicate of #1641

vojtechkral, 2019-01-14:
@Mirarkitty yes, more specifically, wxWidgets on top of GTK. It's quite quirky.

Mirarkitty, 2019-02-05:
Same behaviour in alpha 5.

bubnikv, 2019-03-02:
The HiDPI issues shall be fixed in the upcoming alpha7. Closing.

closed 2019-03-02 by bubnikv

Mirarkitty, 2019-03-16:
Confirmed, fixed in alpha 7. Thank you