prusa3d/PrusaSlicer #1645: Slic3r PE alpha 3 - Object Manipulation Box Values Incorrect

cosmith71, 2019-01-12:
### Version
Slic3rPE 1.42.0-alpha3+win64

### Operating system type + version
Windows 10 Home Build 1803

### Behavior
* _Describe the problem_

The size reported in the scaling box do not match the size of the object

* _Steps needed to reproduce the problem_

Load any object and observe the sizes in the scaling boxes and the actual size.

* _Expected Results_

These should match.

* _Actual Results_

 They don't match.  This seems to be new since alpha-2.

[image]

[image]

FidelCapo, 2019-01-14:
I can not reproduce this issue. Can you please send 3MF file? Thank you.

FidelCapo, 2019-01-14:
Thank you for steps to reproduce. Now I can reproduce it for first selection. When I reselect object values seems to be ok.

Created JIRA 764

cosmith71, 2019-01-14:
Same here.  Thanks!

FidelCapo, 2019-01-15:
Should be fixed with next release.

closed 2019-01-15 by FidelCapo