prusa3d/PrusaSlicer #1646: Slicer 1.42.0 alpha 2 and 3 - Z position reflects center of object and not bottom

cosmith71, 2019-01-12:
### Version
Slic3rPE 1.42.0-alpha3+win64

### Operating system type + version
Windows 10 Home Build 1803

### Behavior
I don't know if this is working as designed or not, but when an object is loaded, its Z position is shown as half its height.  It may be more useful to have the z position reflect the lowest part of the object so that 0 results in the object's bottom touching the build plate.

* _Expected Results_
Z coordinate determined by the bottom of the object so that a value of zero results in the object touching the build plate.

* _Actual Results_
 The z coordinate seems to be based on the center of the object.

[image]

Rowbotronics, 2019-01-12:
Yeah, this is consistent with the other axes (X and Y are centered in the middle) but it is a bit confusing and counter-intuitive on the Z axis.

JesusFreke, 2019-01-24:
I ran into this as well. Having 0 be the middle makes it very annoying to accurately position modifiers. I often want to place a modifier that affects a specific layer or a few contiguous layers. This is easy with 0 = bottom by looking at the layer view and making a note of the z height of the relevant layers. With 0 = middle, I have to actually do... *gasp*... math, in order to figure out the exact placement to use.

bubnikv, 2019-05-22:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/1802

closed 2026-09-29 by prusaslicer-bot[bot]