prusa3d/PrusaSlicer #1647: [Request] Setting to set a custom configuration (profiles) folder

photog0411, 2019-01-12:
### Version
any

### Operating system type + version
all

### Behavior

[Request] Setting to set a custom configuration (profiles) folder

_Is this a new feature request?_

yes

lukasmatena, 2019-01-13:
This has been implemented years ago, the custom configuration folder can be set by using the `--datadir` command line parameter. That is, calling `./slic3r-gui --datadir /home/user/my_config_directory` (Linux example) will do what you need.

photog0411, 2019-01-13:
How do I do that on Windows?

lukasmatena, 2019-01-13:
I assume that the command line approach will be the same, that is, cd into directory with the executable and call `slic3r.exe --datadir C:\whatever`. Creating a shortcut that always passes the parameter would be more user-friendly for most: https://www.youtube.com/watch?v=dIfpnqgh1HY.

bubnikv, 2019-01-14:
--datadir should solve the request. Closing.

closed 2019-01-14 by bubnikv