prusa3d/PrusaSlicer #1648: 1.42.0-alpha3: buttons/objects in plater area does nothing

Mirarkitty, 2019-01-13:
### Version
Slic3rPE-1.42.0-alpha3+linux64-full-201901111425.AppImage

### Behavior
* importing one STL object or several STL objects
* clicking any button does not perform any action related to the button
* same behaviour as if the click/drag would have been without the button

* can also not select items (any click deselects)
* not sure if there is a right click menu anymore, but that also doesn't open

Actions that can be performed are rotating the board (left drag), deselect (left click), panning the board (right click). No matter if on a button or on an object.

lukasmatena, 2019-01-13:
Thanks for reporting. I assume that by "button" you mean all buttons from both the top and left toolbars and the 3D / Preview switch, and that neither of these work. We will look into it, but I think a bit more info could be helpful:
- does it work correctly in 1.42-alpha2?
- does hovering over an object change its color, or is it also broken?
- do keyboard shortcuts work correctly? (Select an object through the object list, move the mouse into plater area and press R - rotate gizmo should be activated.)
- is there something special about your system settings (HiDPI screen etc)?
- could you reveal what Linux distro you are running and ideally provide all info that Slic3r shows on clicking Help -> System Info?

It is possible that @enricoturri1966 (or someone else) will see the cause rightaway, but if that's not the case, anwers to these questions might give more clues.

Mirarkitty, 2019-01-15:
The bug is present on alpha1.

bubnikv, 2019-01-15:
> Hovering changes the colour of the object and also the button hover
graphics works/activates.

> Where is the object list? Select all selects the object and R works.

> I have a HiDPI screen (3840x2160 on 15"). I use fvwm2 as window manager
on Mint.

Sorry, I cannot make sense of it. Maybe a screenshot would help?

Do you have an issue with the About dialog? With the main window?

Mirarkitty, 2019-01-17:
Clicking on anything in the main window, buttons or objects, does not work since alpha 1. Was that unclear?

I was asked for the output for system info.

bubnikv, 2019-01-17:
> I was asked for the output for system info.
> I'm not sure how a screenshot would help for information on interactions.

That would help us to get an idea about your graphics card and graphics driver.

heczis, 2019-01-28:
The same behavior here, with Slic3rPE-1.42.0-alpha3+linux64-full-201901111425.AppImage on
ArchLinux (kernel 4.20.4) + PekWM + xf86-video-intel and mesa graphics drivers
Attaching a screen recording that shows the unsuccesful use.
[slic3r.zip](https://github.com/prusa3d/Slic3r/files/2802141/slic3r.zip)

heczis, 2019-01-28:
And, BTW, I observe the same in version 1.41.2 (with the plater-located buttons - see the attached fig.).
[image]
However, the incorrect behavior is observed in AppImage versions only. The natively compiled versions or the one installed from official repositories (another machine/distro) work well.

bubnikv, 2019-01-28:
> I have a HiDPI screen but I don't see how this is relevant

That may be quite relevant.

@heczis

> However, the incorrect behavior is observed in AppImage versions only. The natively compiled versions or the one installed from official repositories (another machine/distro) work well.

We don't see any reason of what may make the difference. Does your compiled Slic3r PE links against GTK2 or GTK3? What wxWidgets library does it link against?

Mirarkitty, 2019-02-05:
Same behaviour in alpha 5 appimage.

Mirarkitty, 2019-02-05:
I would build my own and see if it works better, but the build instructions for Linux are messed up and I can't spend more time on this.

bubnikv, 2019-02-05:
>  but the build instructions for Linux are messed up

What exactly is messed up?

Mirarkitty, 2019-02-05:
> > but the build instructions for Linux are messed up
> 
> What exactly is messed up?

That seems a different bug, but I ran into three issues: the clone instructions are for Slic3r, not Slic3rPE since you link to that manual, "git pull" is not a valid command and running the perl scripts ends with "In the next step, you need to build the [...] library". A fourth issue might or might not be that it can't build with the version of wxWidgets I can install.

vojtechkral, 2019-02-05:
@Mirarkitty I'm not sure what doc you're referring to, but the current docs on how to build on Linux are [here](https://github.com/prusa3d/Slic3r/blob/801cf8451736f7e26b534420241c174d7269ae1e/doc/How%20to%20build%20-%20Linux%20et%20al.md). You shouldn't need to run any perl scripts. As for wxWidgets, the problem is dicussed in #1779 , basically, this can be done with `-DSLIC3R_WX_STABLE=1` (I'll add it to the doc tomorrow-ish).

Mirarkitty, 2019-02-09:
The linux document is linked from https://github.com/prusa3d/Slic3r/blob/801cf8451736f7e26b534420241c174d7269ae1e/README.md to https://github.com/slic3r/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux ...
[image]

vojtechkral, 2019-02-09:
Eh we should probably just delete or replace that readme file...

Mirarkitty, 2019-02-12:
I did get stuck on not having a modern enough Boost library now (Boost.Process is missing). I have a few year old dist that was tweaked to work on my hardware, and I can't risk upgrading that... :p

Mirarkitty, 2019-02-12:
Is there anything I can test though? :p

bubnikv, 2019-03-02:
There is an alpha7 coming with improved HiDPI support, but I am not sure whether it will fix the GTK2 vs GTK3 issues.

bubnikv, 2019-03-02:
We may one day need to precompile two builds for Linux, one against GTK2 and the other against GTK3.

heczis, 2019-03-03:
> ... whether it will fix the GTK2 vs GTK3 issues.

I ran 1.42.0.alpha7.r1 compiled against GTK2 and the plater-area buttons work now.

Mirarkitty, 2019-03-16:
Confirmed, alpha 7 works great. Thank you

closed 2019-03-25 by vojtechkral