prusa3d/PrusaSlicer #1650: 1.42.0-alpha3: STL interpreted differently than other viewers

TLA8, 2019-01-13:
### Version
1.42.0-alpha3+win64

### Operating system type + version
Windows 10 Version 1803

### Behavior
The same .stl file(s) opened in 1.42.0-alpha differs from what previous 1.41.2 Slic3r, or Cura/Tinkercad/Fusion360/Thingiverse/Paint 3D (or other similar viewers) are displaying.

1.42.0-alpha shows extra layers and blocks the hole in this thing:
[https://www.thingiverse.com/thing:3044013](url)
The error stays also after slicing so the printed part needs some extra work to clean up. I could get an somehow improved version by importing+exporting the file in Tinkercad.

This particular .stl consists of two pieces (possibly some related errors) and somehow the latest Slic3r combines that differently than other softwares.
[image]

Sebastianv650, 2019-01-13:
The real problem isn't Slic3r, but all the CAD programs which are producing faulty stl files. All the blame goes to the slic3r then due to the users which are not aware of that issue. In this special case, it might also be a user error from the creator of the file, as it looks like he just slamed together 2 stls without caring to create a valid mesh out of them.

I did a manual repair in Meshmixer, here is the fixed file which now should work with every slicer. Just rename it from .txt to .stl:
[Squash_feet_2041_fixed.txt](https://github.com/prusa3d/Slic3r/files/2753092/Squash_feet_2041_fixed.txt)

TLA8, 2019-01-13:
Thanks @Sebastianv650  for the comment and fix for that file. 

Yes I fully understand that the particular file had some errors, however I can imagine there are a lot of other files with similar issues out there and especially for newcomers the current versions behavior might be too strict and make it too inviting to use other slicers. 

I do not know the actual changes affecting the .stl handling but would it be possible to make it more compatible to these kind of errors? Similarly as in Slic3rPE-1.41.2.

bubnikv, 2019-01-13:
It looks like same of the changes done to the 1.41.2 did not make it into
the 1.41. I have to find the differences in the code, but due to some
internal changes, the review process will be quite lenghty .

bubnikv, 2019-01-14:
The issue was caused by Slic3r trying to triangulate a hole or a crack in the input mesh. The Slic3r 1.41.2 stable, and now with the following commit cf8e4fd7b0c8374c98daffa59e46ae2d5c14651a also the upcoming Slic3r 1.42.0-alpha4 will not try to close the hole on the mesh, but during the slicing.

Closing.

closed 2019-01-14 by bubnikv