prusa3d/PrusaSlicer #1651: [Request] Add [layer_num] and [layer_z] to Custom Tool change G-code

wjsteele, 2019-01-13:
Currently, if you try to use [layer_num] or [layer_z] placeholders in the custom "Tool change G-code", it fails with:

!!!!! Failed to process the custom G-code template toolchange_gcode
Parsing error at line 1: Variable does not exist
M117 Changing Tool on layer [layer_num] at [layer_z]mm
                     ^
!!!!! End of an error report for the custom G-code template toolchange_gcode

It really doesn't make sense that these two values would not be available in this block of code, as some tool changers require the Z axis to be altered.  The normal gcode that follows it does not return back to the correct Z position if you move it within this block.  The layer change g-code does not get placed when a tool change on the same line occurs, so it's not an option... and the filament change only occurs after the layer has changed, which is too late.

Am I missing something?

bubnikv, 2019-01-13:
BTW, is it with the single extruder multi material mode enabled?

wjsteele, 2019-01-13:
Yes... single extruder in my case.

wjsteele, 2019-01-13:
Actually, I just tried it with it unchecked and it failed as well.

rainersc, 2019-01-26:
i am just experiencing the same problem when trying to implement a tool change gcode that needs to use layer_num variable
i am trying to change the extruder temperature when changing to a new filament:

`{if [layer_num] == 1}M109 S[first_layer_temperature[next_extruder]]`
`{if [layer_num] > 1}M109 S[temperature[next_extruder]]`
`M117 change to [next_extruder]`
`M600`

this leads to slic3r (1.42..0.alpha3)  giving me the following error:

`!!!!! Failed to process the custom G-code template toolchange_gcode`
`Parsing error at line 1. `
`{if [layer_num] == 1}M109 S[first_layer_temperature[next_extruder]]`
`       ^`
`!!!!! End of an error report for the custom G-code template toolchange_gcode
`
(The "arrrow" points to "[layer_num]"

I am using a mk3 profile without mmu but configured to support multiple extruders for manual color changes.

bubnikv, 2019-01-26:
Please note that the Slic3r PE macro language is different (better) from
the upstream slic3r
https://github.com/prusa3d/Slic3r/wiki/Slic3r-Prusa-Edition-Macro-Language

wjsteele, 2019-01-26:
Yes, understood, but the page you linked too clearly documents the issue... Both layer variables are only available in the custom layer change gcode. Our request is to extend them to the custom tool change gcode as well.

rainersc, 2019-01-28:
When invoking a tool change, usually a filament change is the reason. as different filaments require a different temperature, we need to handle this in the tool change gcode.
but the filament temperature is usually different for the first layer, that is why the 2 layer variables are required to be available in the tool change gcode..

bubnikv, 2019-01-29:
Implemented with aaaffd7f90d843d585e624a39ea24fe73c9d9ac1
Closing

closed 2019-01-29 by bubnikv