prusa3d/PrusaSlicer #1652: dynamic layerheight several fatal problems

akust, 2019-01-13:
### Version
1.42.0-alpha2+
### Operating system type + version
MacOS Sierra 10.12.6

### Behavior
* _Describe the problem_
several fatal errors in the dynamic leyerheight editor:
1. layerhight width line not shown at all without clicking on it
2. any edit gets reset after releasing and moving the mouse!
3. bad contrast: dark green bg vs. blue line

* _Steps needed to reproduce the problem_
my setting right now: '0.10mm DETAIL...'
1. - should be obvious immediately: there is no line
2. try to (in my case) thicken layerhight anywhere, release mouse - bam: all reset
3. ok, not too bad _if_ the line is visible but kind of too dark anyway (i hope you agree)
* _Expected Results_
..
* _Actual Results_
dynamic layerhight literally unusable
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

[image]

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

akust, 2019-01-13:
just updated to alpha3 - same problem

lukasmatena, 2019-01-13:
Closing as a duplicate of #1499 (which is itself redundant).

closed 2019-01-13 by lukasmatena