prusa3d/PrusaSlicer #1659: [v1.42.0-alpha3] Supports around walls and mid-air

jurriaan, 2019-01-14:
### Version
Slic3r v1.42.0-alpha3 

### Operating system type + version
x86_64 Linux (Arch)

### Behavior
Load config and stl and enable supports, you will see support structures in places they don't belong

[image]

#### STL/Config (.ZIP) where problem occurs
[slic3r-bug.zip](https://github.com/prusa3d/Slic3r/files/2757232/slic3r-bug.zip)

phowson, 2020-12-11:
I am also experiencing this issue . 
[image]

Version: 2.2.0+win64
Build: PrusaSlicer-2.2.0+win64-202003211145
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19041), 64-bit edition

Happens on certain specific STLs, for example with these gears, around the points I see supports in mid air. Attached project file, so hopefully you can reproduce

[Mid air supports.zip](https://github.com/prusa3d/PrusaSlicer/files/5678442/Mid.air.supports.zip)

bubnikv, 2021-02-24:
Fixed on current master.

closed 2021-02-24 by bubnikv