prusa3d/PrusaSlicer #1660: Repetier integration and -o argument

Itox001, 2019-01-14:
Using Repetier Host with the latest slic3r pe 1.42 alpha 3, I get a g-code not found error.

Viewing the logs, I see that Repetier calls slic3r with the --o "composition.gcode" argument, which slic3r does not recognize according to the following message:

> Warning: unknown option --o

I think that that what Slic3r wants is the --output argument instead of -o. Was this changed recently?

The 1.41.2 version does not have such problem.

bubnikv, 2019-01-15:
Please see https://github.com/prusa3d/Slic3r/issues/1582

> The command line parser and logic written in C++ that was taken from
upstream was non functional, now the upstream C++ command line processing
evolved a lot, we have to merge it.

So basically the command line processing has yet to be finished for command
line slicing.

bubnikv, 2019-01-15:
> Understood @bukniv, thanks for the quick answer.
> Hopefully at least that argument will be implemented soon, since slicing
> from repetier is a vital part of my workflow.
> With the new UI Slic3r is getting much easier to work with so it's not as
> painful as it once was, but still some extra steps to import the g-code
> into the host.
>
> Thanks for the good job!

> With the new UI Slic3r is getting much easier to work with so it's not as
painful as it once was,

What is the functionality in Repetier Host, that you are missing in the
current 1.42.0-alpha3?

> but still some extra steps to import the g-code into the host.

Is it just the "send to repetier-host" function, that you are missing?

Itox001, 2019-01-16:
> What is the functionality in Repetier Host, that you are missing in the
current 1.42.0-alpha3?

It´s mainly the Host functions, connect to the printer and send g-code, display and graph the temps, etc. It used to be also object positioning, scaling and rotation, becuase slic3r's UI used to be horrible, but I realize that that has now changed and now Slic3r's UI is more powerful in that aspect. 

One thing that is actually missing is the parallel/perspective view toggle. Its useful for when you want to position things precisely (parallel) or when you want it to look natural (perspective). Also, I miss the "Center object" button.

> Is it just the "send to repetier-host" function, that you are missing?

That would be very useful. I just found the "Watch List" function of Repetier Host, which basically lists all gcode files in a directory, sets flags for new/changed files and has a Load button to quickly import them. So that helps relieve the inconvenience a little...

The awkward thing imo is having to export/import gcode. Either the "export to repetier" button or the watch list help with that. 

BTW I played some with the new UI today, and congratulations, it´s much much better looking and easier to use than it was. I miss some things from repetier host such as the model lighting and color, backgroud color and things like that, but I realize that that is only becuase of the familiarity.

bubnikv, 2019-01-16:
The model color is there, just click at the color box at the filament
selector combo.

bubnikv, 2019-03-14:
The 1.42.0-beta command line processing is based on the upstream code by @alexrj and @lordofhyphens, thanks to them.
The command line format is described by https://manual.slic3r.org/advanced/command-line
with some exceptions:
- instead of --help-options we have --help-fff and --help-sla
- we have an additional parameter --loglevel
- we support --export-sla, not --export-sla-svg or --export-svg
- we don't support --cut-grid --cut-x --cut-y --autosave

We are not going to implement a "repetier compatibility" backward mode. It is up to the repetier guys to update their software. As we have adapted the command line interface of the upcoming C++ only upstream slic3r, repetier will have to be adapted for the upstream slic3r as well.

bubnikv, 2019-03-14:
@repetier @repetierhost guys, are you listening?

repetier, 2019-03-15:
Now you mention me:-)
So -o has become --export-gcode
Will add that with a new version selector in next release.

bubnikv, 2019-03-16:
> So -o has become --export-gcode

Not exactly

The parameter -o (or --output) works the same way as before. The old slic3r would start in GUI mode depending on the --gui or --nogui switches. The new slic3r will start GUI if it does not receive any action command (action command being for example --slice, --export-gcode or --export-sla).

So you should call
slic3r input.amf --export-gcode --output output.gcode

repetier, 2019-03-16:
Understood. Thank you for clarification. Will add it for next update.

bubnikv, 2019-03-28:
I am closing this issue at slic3r PE tracker as the ball is now at repetier.

closed 2019-03-28 by bubnikv