prusa3d/PrusaSlicer #1661: Alpha 3 MMU problem.

RJ-Make, 2019-01-14:
### Version
Version: 1.42.0-alpha3+win64
Build: Slic3rPE-1.42.0-alpha3+win64-full-201901111426
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 17134), 64-bit edition

### Behavior
When sliced in A3, sent to the printer direct via OctoPrint, the picture below is what takes place (I moved the extruder up, but normally just sits on the last layer oozing plastic).

OcotoPrint ack's like the print is paused, but the only possible response within OctoPrint is to cancel or Restart.

[Tesla_Base_0.25mm_ABS_MK3MMU2.zip](https://github.com/prusa3d/Slic3r/files/2757445/Tesla_Base_0.25mm_ABS_MK3MMU2.zip)
  
[image]

bubnikv, 2019-01-15:
Does it print directly without OctoPrint?

There is a
M1 S10
at line 201 of your G-code, which instructs the printer firmware to wait
for 10 seconds to give you a bit time to verify, whether all the filaments
were loaded and primed correctly.

By the way, we do not recommend to do the priming at the edge of the print
bed for the MMU2. This only happens because you enabled the checkbox "Prime
all printing extruders" at the Print Settings -> Multiple Extruders page.

RJ-Make, 2019-01-15:
Thank You, I didn't notice that setting. I checked and the non-Alpha profiles have that setting unticked. Not exactly sure why I turned it on in my Alpha testing but that was indeed the entire problem. 

Not sure what the M1 S10 is all about. I simply unticked the "Prime all printing extruders" and everything is now working as it did in 1.41.2+win64 regarding the MMU2.

closed 2019-01-15 by RJ-Make