prusa3d/PrusaSlicer #1662: Complete Individual Objects should print from front to back

timur-tabi, 2019-01-15:
### Version
1.41

### Operating system type + version
Mac OS 10.13

### Behavior
When printing individual objects, the objects should be printed from front to back, otherwise tall objects printed on the back of the plate will collide with the x-axis rods when the printer tries to print objects on the front of the plate.

wavexx, 2019-10-26:
Import/instance order should be essentially ignored in both cases.

Whenever "sequential printing/complete individual objects" is enabled the only concern should be extruder collision. Objects should be ordered by Y, then printed in a zigzag pattern (similar how the leveling is performed) so that travels from objects on the same Y lane do not cause a collision too.

When sequential printing is disabled, the primary concern should be minimal travel distance, which could change depending on the layer.

I don't think there should be any reason to set a manual order.

bubnikv, 2020-02-04:
> Objects should be ordered by Y, then printed in a zigzag pattern (similar how the leveling is performed) so that travels from objects on the same Y lane do not cause a collision too.

That applies to the i3 kinematics. For other machines the optimal ordering may be different.

Anyway, the upcoming PrusaSlicer 2.2.0-alpha4 will print sequentially in the order displayed at the object list. Also the object list will support drag & drop to reorder. This does not solve the issue of auto-placement, but at least it gives one a way to optimize the sequential order manually.

Jaxx005, 2021-05-28:
The details of the scenario I faced are at the end, but here's a summary. Sorry if not in correct place, but this seemed the most relevant entry. Noobie here.

[FEATURE REQUEST] Add option to order printing from front to back row (and by different methods?)
The front-to-back print order would not always be feasible, but I agree it seems a good solution to try first: it's fine for instances that are narrow enough to remain clear (in front) of the horizontal X axis, and especially those that are low enough to avoid everything!

[FEATURE REQUEST] Avoid collision with wiring harness (!) at rear of print head
The current auto print order seems to ignore the wiring harness (which is relatively delicate compared to the print head)? Certainly caused an issue in my print case (see below). Printing from front row to back row would avoid this (in many cases) as the bed behind is yet to be printed upon.

[FEATURE REQUEST]  Show print head & wiring harness in toolpath 'playback'
This might be overkill, but certainly useful IMO... and it might be simple to implement from the CAD models?

Example scenario...
Below is the detail I posted under Comments on PrusaSlicer KB (as "LeeSheridanTech").  

See also screenshots...
[image]
[image]

I found that when printing 6 instances (L75 x W17 x H12, on MK3S) and using the auto-layout, the print order made no sense to me (eg. switching between rows and columns of instances). This led to instances directly in front of previous ones.  Which in turn, would have lead to the wiring harness (at the rear of the hotend) colliding with those previous instances.   Perhaps collisions with wiring is something to avoid even more so than with the mechanical parts around the hotend? Layout optimisation is a knotty problem, but at least for this case scenario (!) it would seem that working from the front is a good solution to try first: there is nothing in front of the print head.  This works fine for 'slim' objects, as the horizontal axis can be kept clear of previous instances on the same row.  Describing the problem is itself 'knotty', so I'll stop there in the hope that it kind of makes sense!? At least for multiple-instances of the same model?   Perhaps 'optimsation' has to be left to the user - in which case an option to "order by row then column" and vice versa, would be useful imo  ;) Just had a thought, would it be easy to add an option to show the moving parts in the slicer 'replay'?  Maybe this is all overkill. mmh.

Excellent work people. Lovin' this software. Many thanks! :)

lukasmatena, 2025-02-18:
Arrange and crash detection for sequential printing was improved in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). I believe it solves this issue. Your feedback is welcome. Closing.

closed 2025-02-18 by lukasmatena