prusa3d/PrusaSlicer #1664: Unexpected bridge infill?

eblot, 2019-01-15:
### Version
1.42.0-alpha3+
(BTW, the "about" dialog closes whenever it is clicked, w/o a way to copy/paste the version. Is this the expected behaviour?

### Operating system type + version
macOS 10.14.2

### Behavior
* Unexpected bridge infill is created (see attached snapshot)
* Import STL model, slice with "0.20mm FAST MK3" system setting, nothing else.
* Expected Results: no bridge (I'm adding a snapshot from Simplify3D for comparison)
* Actual Results: a layer with bridge infill is created. I do not get why.

#### STL/Config (.ZIP) where problem occurs
[stl-config.zip](https://github.com/prusa3d/Slic3r/files/2758466/stl-config.zip)
[image]
[image]

bubnikv, 2019-01-15:
This is a duplicate of #1562, #1592, #1614, #1633 and it has been fixed in the current master, the fix will be contained in the following release.

I am keeping this issue open for reference.

FidelCapo, 2019-01-15:
@bubnikv it slicing correctly with 1.42.0-alpha3+31.

@eblot About and System info dialogs should be fixed too.