prusa3d/PrusaSlicer #1666: 1.42.0-alpha3: post processing script not run

Scachi, 2019-01-15:
### Version
Slic3rPE-1.42.0-alpha3+win64-full-201901111426

### Operating system type + version
 Win10 x64 Pro - Version 1803

### Behavior
post-processing script is not run, there is no error message or I don't see one.
it works fine in version Slic3rPE-1.41.2+win64-full-201811221508

I have this in the Post-processing script field:
`d:\slic3r_script.bat;`

content of the .bat file:
```
C:\"Program Files"\Prusa3D\Slic3rPE\perl5.24.0.exe d:\slic3r_my.pl %*
```
.bat and .pl file attached
[ppscript.zip](https://github.com/prusa3d/Slic3r/files/2761178/ppscript.zip)

bubnikv, 2019-01-16:
I suppose it's because

```
#ifdef WIN32

namespace Slic3r {

//FIXME Ignore until we include boost::process
void run_post_process_scripts(const std::string &path, const PrintConfig
&config)
{
}

} // namespace Slic3r

```

Now seriously, support of the post processing scripts was not yet fully
ported from Perl to C++. It currently works on Linux & OSX, but not yet on
windows. We should fix it.,

closed 2019-01-17 by bubnikv

bubnikv, 2019-01-17:
Implemented with  668d374

bubnikv, 2019-01-17:
By the way, the perl scripts may be executed with an optional Perl interpreter.

https://github.com/prusa3d/Slic3r/wiki/Running-Perl-post-processing-scripts-on-Windows-from-Slic3r-PE-1.42-and-newer

leerc, 2019-01-26:
I'm having problems running batch files from the script prompt e:\python\dome.bat. Works in 1.41.2+windows64.
LeeRc

vojtechkral, 2019-01-28:
> I'm having problems

What problems are you having?

bubnikv, 2019-01-28:
it should be fixed in alpha4, see the release log. Closing.