prusa3d/PrusaSlicer #1667: Slic3r shows cryptic filename from Fusion 360's direct STL export

MartinMajewski, 2019-01-15:
### Version
1.42.0 - alpha 3

### Operating system type + version
MacOS 10.14.3

### Behavior
When exporting STL in Autodesk Fusion360 and selecting the output option "Send to 3D Print Utility" the file shown has a cryptic name instead of the body name defined in Fusion 360 (which is the standard output when saving the STL to disk).

[image]
[image]

If for whatever reason there is no way to obtain the body's correct name during the export/opening, we should have to have the option to rename the objects in Slic3r's object browser window. Currently, you have to save the STL, chose a new name and reopen it after removing the current one from the print surface of Slic3r. This is not convenient nor helps it organizing the print job.

MartinMajewski, 2019-01-15:
Furthermore, sending the cryptic name to OctoPrint gives this faulty output:

[image]

bubnikv, 2019-01-15:
what if this cryptic name is generated by fusion 360? what do you advice?

MartinMajewski, 2019-01-15:
> If for whatever reason there is no way to obtain the body's correct name during the export/opening, we should have to have the option to rename the objects in Slic3r's object browser window. [...]

This! :-D

MartinMajewski, 2019-01-15:
Just give use a rename option at the object brower / object list. It is still easier to copy/paste the name from Fusion to Slic3r than going the "Save STL" route in Slic3r - which seems not to be present in the alpha 3?! Please reinsert it as Slic3r is my tool of choice to reorient objects and save them as STL for e.g. releasing them on Thingiverse.

vojtechkral, 2019-01-24:
@MartinMajewski Could you please share an example STL file that has this problem?

YuSanka, 2019-01-31:
Editing of the object/part's name under OSX, using pop-up window, is implemented in https://github.com/prusa3d/Slic3r/commit/411c02fef3efe1329baf1192e2d8645064d93780. It will be a part of the upcoming release.

closed 2019-01-31 by YuSanka