prusa3d/PrusaSlicer #1670: Need a placeholder for the next coordinates after tool change

fpikus, 2019-01-16:
### Version
1.41.2

### Operating system type + version
Windows 10 64-bit

### Behavior
I'd like to be able to do this in the tool change script, it gives the best results for ooze control in my printer (dual bowden extruders). 
G1 E-8 F15000                     ; retract old extruder slowly
G91 		        ; relative positioning
G1 Z0.5 F900                       ; lift nozzle
G90 		        ; absolute positioning
G1 E-27 F[retract_speed]   ; retract old extruder
T[next_extruder]                 ; change extruders
G91 		        ; relative positioning
G1 Z-0.5 F900                     ; lower nozzle
G90 		        ; absolute positioning
G1 E35 F[retract_speed]    ; prime new extruder
G92 E0	                         ; Zero the extruded length

There is only one problem: After the extruder change (T1, say), the new extruder is primed at the last coordinates of the old extruder. Say, I print with red and blue. I switch from blue to red, the red extruder goes where blue was last when it lifted and leaves a small red dot there. Same when I change to blue. 

For comparison, S3D has the same script except there I can do this:
G1 X[next_position_x] Y[next_position_y] F[travel_speed]    ; move to next printing location

Now the red extruder is primed at the location where red extrusion begins, not where blue ended. No mixing filaments.

_Is this a new feature request?_
Yes

#### STL/Config (.ZIP) where problem occurs

bubnikv, 2019-01-16:
This may be a bit more involved to implement. Do you know any slicer that
supports such a functionality?

fpikus, 2019-01-16:
Yes, Simplify3D does: [next_position_x], [next_position_y].

NGC869, 2021-10-06:
Hi,
[next_position] placeholders would be also extremely useful when used as last command in the _Start Gcode Script_, because often is better to first bring the nozzle in the start X/Y position and then lower the nozzle (or rise the bed) in the Z direction.

Actually I'm using this kind of instruction: 
_G1 F12000 X[first_layer_print_min_0] Y[first_layer_print_min_1] Z{first_layer_height + 2.0}_
but these coordinates are related the the imaginary rectangle boundaries of the printed part, not the real start print coordinates, so leaving a dot of polymer outside the real print.

Thank you,

foreachthing, 2021-10-06:
@NGC869 there you go (Cura-Move); Use the python script 
https://github.com/foreachthing/Slic3rPostProcessing
Still working on documentation but should easy to understand from the content of the script.

NGC869, 2022-12-27:
Hi,
any chance to see the **[next_position_x/y/z], [first_position_x/y/z], [last_position_x/y/z] ** placeholders in the next release?

Thank you!

closed 2026-09-29 by prusaslicer-bot[bot]