prusa3d/PrusaSlicer #1672: [Feature Request] Ultimaker 3 support

Salamandar, 2019-01-16:
I have an Ultimaker 3 & Ultimaker 3 Extended at work, and we're currently using Cura to slice + send the file.
I'm trying to use Slic3r, but it's kinda difficult, and some modifications are needed for a perfect compatibility.

# What's needed : (probably not exhaustive for now)

- [ ] Default profile with dimensions and some default parameters
- [ ] Ultimakers are waiting for a header **at the very top** of the GCode. They documented it on their forum, i'll paste here with descriptions. It can't be put in "before start custom gcode" as Slic3r adds its own header… **That's actually the main issue here.**
- [ ] Ability to put in the header infos such as dimensions, duration, volume needed.
- [ ] (actually a feature request) "wipe tower"/"purge extruder"  at start : the Ultimaker extrudes like 3cm of filament in a tower before starting the print, kind of like the skirt. It can be added in the custom GCode, but a check box for this would be great.

I don't really know the source code of Slic3r, but I'm ready to get my hands dirty if i'm not alone on this ;)

Salamandar, 2019-01-16:
Here is the full Cura/Ultimaker header (with wipe tower and useless things).
It's fully documented here : https://community.ultimaker.com/topic/15555-inside-the-ultimaker-3-day-1-gcode/
I tried to put this in the "custom gcode/Start G-Code", but it seems the printer is waiting for it at the very beginning of the file… A simple manual post-edit of the file corrects this.
```
;START_OF_HEADER
;HEADER_VERSION:0.1
;FLAVOR:Griffin
;GENERATOR.NAME:Cura_SteamEngine
;GENERATOR.VERSION:3.6.0
;GENERATOR.BUILD_DATE:2018-11-12
;TARGET_MACHINE.NAME:Ultimaker 3 Extended
;EXTRUDER_TRAIN.0.INITIAL_TEMPERATURE:205
;EXTRUDER_TRAIN.0.MATERIAL.VOLUME_USED:11130
;EXTRUDER_TRAIN.0.MATERIAL.GUID:506c9f0d-e3aa-4bd4-b2d2-23e2425b1aa9
;EXTRUDER_TRAIN.0.NOZZLE.DIAMETER:0.4
;EXTRUDER_TRAIN.0.NOZZLE.NAME:AA 0.4
;BUILD_PLATE.TYPE:glass
;BUILD_PLATE.INITIAL_TEMPERATURE:60
;PRINT.TIME:7983
;PRINT.SIZE.MIN.X:9
;PRINT.SIZE.MIN.Y:6
;PRINT.SIZE.MIN.Z:0.27
;PRINT.SIZE.MAX.X:146.314
;PRINT.SIZE.MAX.Y:126.056
;PRINT.SIZE.MAX.Z:47.97
;END_OF_HEADER
;Generated with Cura_SteamEngine 3.6.0
T0
M82 ;absolute extrusion mode

G92 E0
M109 S205
G0 F15000 X9 Y6 Z2
G280
G1 F1500 E-6.5
;LAYER_COUNT:320
;LAYER:0
M107
M204 S625
M205 X6 Y6
;MESH:3DBenchy.stl
G0 F3750 X125.327 Y111.545 Z0.27
M204 S500
M205 X5 Y5
;TYPE:WALL-INNER
```

bubnikv, 2019-01-16:
As far as I understand it, to get your Ultimaker printer to print, you likely just need to create the print/filament/printer profiles and set the correct start/end G-code, right?

> Ability to put in the header relevant infos such as dimensions, duration, volume needed.

Is this necessary for the Ultimaker to print?

We may do some very minimal changes initially to make Slic3r work with Ultimaker, but if extensive changes are needed, we would leave it to the community.

Salamandar, 2019-01-16:
> As far as I understand it, to get your Ultimaker printer to print, you likely just need to create the print/filament/printer profiles and set the correct start/end G-code, right?

Not only. The start GCode puts the header **after** the Slic3r header, but Ultimaker needs it **before**.

> > Ability to put in the header relevant infos such as dimensions, duration, volume needed.
> 
> Is this necessary for the Ultimaker to print?

No, it's completely optional. But without it, the time estimation displayed on the screen is completely false.
> 
> We may do some very minimal changes initially to make Slic3r work with Ultimaker, but if extensive changes are needed, we would leave it to the community.

Would the header issue be fixed by a new GCode flavour ?

bubnikv, 2019-01-16:
>  Would the header issue be fixed by a new GCode flavour ?

That is an implementation option. I know the "Gremlin" G-code flavor was
derived from Marlin, so adding a "Gremlin" G-code flavor may make sense.

capi92, 2020-04-13:
> > As far as I understand it, to get your Ultimaker printer to print, you likely just need to create the print/filament/printer profiles and set the correct start/end G-code, right?
> 
> Not only. The start GCode puts the header **after** the Slic3r header, but Ultimaker needs it **before**.
This can be achieved via post processing script.

Salamandar, 2020-04-13:
That's actually what I did, but that's not an amazing solution ;)

Salamandar, 2024-05-04:
Unfortunately I don't have access to an Ultimaker anymore, so I can't really test that :(
I guess that until someone else have an issue, you can close it.

wrennnnnn, 2024-05-06:
Works for me with a slightly modified start gcode and dimensions, however on both my UM3E and my S5 remaining time does not work with @ansonl's profile. Additionally, the gcode needs to be postprocessed, and purge towers do not work (as the ultimaker line does not support relative extrusion.

Prints do look good, although perhaps some special handling of color changes needs to be done, as with the default settings there is a lot of stringing on the first couple layers. Maybe a 'print this object first' command in the slicing step would be ideal as a band-aid, so we can make custom purge towers.

ansonl, 2024-05-07:
@wrennnnnn Ultimaker S-line seems to use the estimated print time in the Griffin gcode header and Prusaslicer does not have access to the estimated print time when adding the start gcode so that is why the remaining time might not work.

wrennnnnn, 2024-05-14:
@ansonl ahh, in that case i will simply add it to my postprocess gcode, thank you!

Attached is the postprocessing gcode i've made. All it requires is the default filename, and it converts (messily) the days (if present), hours, and minutes in the SLIC3R_PP_OUTPUT_NAME environment variable to a delta time format, then to an integer, then to a string (to remove trailing zeros.), and also removes the first line (as it will not print with it) It also only searches for a remaining time of 0, for futureproofing.
[ultimakerpython.zip](https://github.com/prusa3d/PrusaSlicer/files/15301149/ultimakerpython.zip)

robinkruyt, 2024-05-21:
I’ve tried using this profile with an original Ultimaker 3. The printer accepts the sliced model fine (after changing the bed size) but when printing almost no filament seems to be extruded. What should be tweaked to correct that?

ansonl, 2024-05-21:
Is the printer the S3? I'm not sure if the 3 and S3 are on different firmwares.
You should check if the generate gcode is in absolute or relative E position mode. Ultimaker firmware only supports absolute extrusion.

wrennnnnn, 2024-05-22:
@robinkruyt works fine on my um3 extended, are you sure it's not a hardware issue? the extruders on the UM line are not very well-regarded.

robinkruyt, 2024-05-23:
Hmm. While I agree with you, it does print fine when sliced with Cura instead.

Isaac-dot-py, 2026-08-25:
@wrencrt Would you mind sharing your printer profile for your ultimaker 3 extended? I recently have come into possession of an ultimaker 3 extended and I'm trying to figure out how to use it with prusaslicer instead of with cura. Thanks in advance!