prusa3d/PrusaSlicer #1674: [Feature Request] layer thickness editing by numerical input

photog0411, 2019-01-16:
Being able to specify a layer range and thickness of layers would be very useful. The "layer editing" slider thing with the graph works poorly and is not very intuitive.

I need to print a model now with first 15 layers at 0.10mm and rest of model at 0.20mm. But layer editing is not allowing me to do it. 

We need input boxes somewhere with layer range where we can actually specify exact layer heights for range.

Icefire55, 2019-01-17:
You can do it in Settings, Layers. You can change layer height (mm) in Min/Max Z (mm)

thenickdude, 2019-01-18:
Unfortunately the Settings -> Layers menu was dropped in 1.42 alpha:

>https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-alpha1
>
>Layer height table was dropped for now, though we plan to replace it with a more general feature to assign not only a layer height, but also other print parameters to layer spans. Assigning parameters to layer spans is implemented in Simplify3D through processes. While Slic3r supports the same feature through modifier meshes (slabs), having the option to define the parameters per layer spans table would be nice #495.

bubnikv, 2020-02-05:
The feature has been re-implemented in PrusaSlicer 2.1. Closing.

closed 2020-02-05 by bubnikv