prusa3d/PrusaSlicer #1676: Slic3r command line feature request

ctilley83, 2019-01-17:
### Version
1.41.2

### Operating system type + version
MacOSX

### Behavior
Invoking Slic3r by the command line does not support 3MF files. Currently slicing by the command line only accepts OBJ, STL, or AMF file types. This could be especially useful if the 3mf file contains metadata for a printer profile in the format as if you selected PLATER -> Export plate as 3mf

1. Process Metadata and load embedded profile
2. Slice model and output to gcode
3. Place Gcode in the specified location.

_Is this a new feature request?_

Feature Request

ctilley83, 2019-01-18:
Can someone explain how the current command line system is structured?

vojtechkral, 2019-01-24:
I think this is being discussed in #1582

bubnikv, 2019-03-14:
The 1.42.0-beta command line processing is based on the upstream code by @alexrj and @lordofhyphens, thanks to them.
The command line format is described by https://manual.slic3r.org/advanced/command-line
with some exceptions:
- instead of --help-options we have --help-fff and --help-sla
- we have an additional parameter --loglevel
- we support --export-sla, not --export-sla-svg or --export-svg
- we don't support --cut-grid --cut-x --cut-y --autosave

bubnikv, 2019-03-14:
1.42.0-beta will support slicing of .3mf and .amf files using the embedded profiles.

closed 2019-03-14 by bubnikv