prusa3d/PrusaSlicer #1677: Negative extra lenght on restarts leads on "holes" in solid layers

tecnopolis-ve, 2019-01-17:
### Version
1.41.1 windows

### Operating system type + version
Windows 10

### Behavior
Setting a negative value to "extra lenght on restart" generate holes in solid surfaces/infills
* Set a negative value to Extra lenght on restart, slice something 
[image]

* A solid surface was expected
[image]

* surface with missing parts
[image]

bubnikv, 2019-01-17:
What did you expect Slic3r to do? Why are you setting a negative value
here? Shall we disable entering a negative value here?

tecnopolis-ve, 2019-01-17:
negatives values are important for minimising blobs after "deretraction" is complete, the correct behavior should be like the first picture (no holes on solid parts) 

the sliced object was a benchy, compare first image with the second, the bug is notable.

metacollin, 2019-01-24:
> negatives values are important for minimising blobs after "deretraction" is complete

No, it's not.  Oozing/blobs are caused by thermal expansion of filament in the process of melting.  Oozing and blobs **always** occur at the end of the print move due to the left over pressure.  Anything done at the start of a print move (as in, a deretraction) will have no effect whatsoever on blobbing/oozing.  

I think what you want is coasting, but may be confused as to how it works, and are perhaps attempting to do it with Slic3r, which as far as I know doesn't support coasting.  

Coasting stops the extrusion move early, and lets the final bit of extra plastic that would normally form the ooze/blob to instead be used as the last part of an extrusion move.  This actually has an effect, as it uses the built up pressure to do something.  

> the correct behavior should be like the first picture (no holes on solid parts)

No, it's not.  The correct behavior is exactly what Slic3r does.  You're telling it to under deretract, so there is 0.4mm of air gap introduced.  Slic3r is simply displaying what is going to happen if you print with this setting.  Displaying the print without gaps would not reflect reality, and therefore be inaccurate.  The gap it shows is the gap you've introduced by making the extruder push nothing but air for the first part of the line, until that extra 0.4mm of air is pushed passed and it can actually begin extruding again.  

Meanwhile, it will still ooze and blob just as much as before, because you're not doing anything about that built up pressure.  You're just letting it ooze anyway, then telling Slic3r to not deretract the filament all the way, which does nothing since retraction already relieved the pressure it could.  

> the sliced object was a benchy, compare first image with the second, the bug is notable

There is no bug, as the second image correctly displays what will be printed negative extra restart distance, and the first one does not.

ceryen, 2019-01-26:
@metacollin You might not believe it, but Simplify3D allows for a negative extra restart distance and people have in fact found it useful for minimizing blobs, particularly at the start of layers.  Google "simplify3d negative extra restart distance" and you'll see such experiences.

How does Simplify3D implement a negative extra restart distance?  Suppose that you have a retract length of 6 mm and an extra restart distance of -0.4 mm.  Retraction length will then be 6 mm and unretract length will be 5.6 mm.  The generated printing traces remain the same, only the position of the filament within the extruder upon unretract is different.

>  Oozing and blobs always occur at the end of the print move due to the left over pressure. 

The negative extra restart distance would affect the end of the print move, which is exactly the scenario that you're talking about.

> Anything done at the start of a print move (as in, a deretraction) will have no effect whatsoever on blobbing/oozing.

Retraction happens at the start of a print move, not deretraction.  Deretraction happens at the end of a print move.

ceryen, 2019-01-26:
Cura also allows for a negative extra restart distance: https://github.com/Ultimaker/Cura/issues/2667

sam2b, 2022-11-13:
Negative values negates the intended purpose of the feature "extra length on restart".

https://reprap.org/wiki/Retraction_Tuning_With_Slic3r#Advanced_Retraction_Tuning_-_Extra_Length_on_Restart
Please read ^

If you are having blobs, have you tried decreasing the retraction amount by 1mm or so?

In my experience on a print with very many retractions, only positive values will solve the problem [what appears to be] under-extrusion causing holes in thin walls.

dbuezas, 2023-10-27:
This trick improved the seam on a specific print for me. (Set to -0.5mm).

closed 2026-09-29 by prusaslicer-bot[bot]