prusa3d/PrusaSlicer #1678: Filenames should be validated/sanitised before sending them to OctoPrint

MartinMajewski, 2019-01-17:
### Version
1.42.0 beta 3

### Operating system type + version
macOS 10.14.3

### Behavior
In addition to my issue submission [#1667](https://github.com/prusa3d/Slic3r/issues/1667) the dialog for sending gcode to OctoPrint should at least have a filename string validation or even a sanitization.

The problem is (as shown by the screenshots) that Slic3r does not put a .gcode file extension behind the filename. Also, it does not check for forbidden characters which are e.g. existing when the filename originates from Fusion 360 directly. Moreover, the filename string is split into hidden parts when a hyphen is present. The user has to move the cursor behind the hyphen to navigate to the remaining portions of the string. This is not immediately noticable which can lead to confusion.

The result of all the above issues is an HTTP 400 error which closes the dialog after hitting the OK button. This is also inconvenient as the entire process of opening the dialog and entering the filename has to be repeated again.

### Proposed solution ###

Check the filename string for

- length (no zero-length strings, maximum length if there is a limit)
- forbidden characters
- the existence of a file extension (.gcode)

and do not break the string into pieces.

The results should be visible to the user immediately as hints underneath the filename or inline and the send button should be deactivated until all criteria are met.

This could be also seen as a Feature-Request

[image]
[image]
[image]
[image]

vojtechkral, 2019-01-24:
Ok, there's a couple of things here:

- The string being broken into parts is a result of a bug in wxWidgets on Mac, this is related to other bugs in settings tabs and will be fixed by simply upgrading wxWidgets to the latest release from december
- Filtering forbidden chars: This is kind of hard to do reliably. Technically speaking, you can generally set whatever characters you want in the filename and it's up to the print host to interpret that. I'm not really sure what "forbidden chars" exactly are and they may be different for each print host
- As for the extension, I'll look into it

foosel, 2019-01-24:
Just for the record, the extension really is a must have here because without any of the valid extensions for gcode (`.gcode`, `.gco`, `.g`, case insensitive IIRC) OctoPrint will deny the upload with the aforementioned 400 Bad Request. Same goes for an empty filename.

Character wise anything UTF-8 should be fine though, but it will be renamed internally to an ASCII representation using slugify.

vojtechkral, 2019-01-24:
@foosel Thanks for clarification

The extension is by-default present and by-default only the filename part without extension is selected.
That is, the extension only goes missing if the user removes it.

However I am noticing that on Mac the whole filename including extension seems to be selected, this seems like a bug.

MartinMajewski, 2019-01-24:
@foosel Danke Dir for your schnelle response!

@vojtechkral You are right. This issue is a chain of unfortunate events. 

1. The entire name gets selected, which leads to having the extension deleted easily with one press of a button.
2. wxWidgets breaks long lines at arbitrary positions (maybe related to characters like scores?!), which in turn looks like the filename actually ends in the middle of the label field.
3. The user can therefore not see that the filename is valid (ugly, but valid) and starts typing, which leads to the consequences of point no. 1.

Therefore, my proposition:

When you select the host type for the extension in the printer settings you immediately know which filename constraints and requirements you have to obey.
While sending to the print host you can check if the filename extension is present.

Now you can provide one of the following outcomes:

- Inform the user that the extension is missing and that the user has to provide one (this would be suitable for any given host type)
- Append the missing file extension during a submission related on the host type selected earlier.

**However, don't do the following:** 
Don't close the dialog! Wait for the host to answer and bring back the dialog with all the user's changes still present. The user should be able to choose if the process should be aborted or not.

Best regards,
Martin

vojtechkral, 2019-01-24:
> Don't close the dialog! Wait for the host to answer and bring back the dialog with all the user's changes still present.

Eh, I don't like this very much. The dialog pops up even before slicing, and the uploading happens later on, the whole thing is asynchronous and happens in the background. The process should be interaction-less. This has been requested by other users.

So far the really only issue I can see is that on Mac the filename is not selected properly, so I'll look into fixing that. Alternatively the filename filed could be split into two fields, a filename and an extension.

vojtechkral, 2019-01-24:
@MartinMajewski Other users have _specifically_ asked for the slicing to be done as part of sending to the printer, see #1173 for an example. And in fact from where I'm standing it does make sense.

If you are concerned about the gcode being ok, you can just hit `Slice now` or `Export Gcode` and only press `Send` once you're happy with the outcome. I don't see a problem here.

If I followed your advice and switched back to when the dialog only pops up after slicing synchronously, then this would be a problem for users who import a model + settings in Slic3r which they now results in valid gcode - they would have to wait for slicing to be done only to pres "ok" on a dialog, which is completely pointless (and in fact people _have_ complained about this before). With the way the upload works now, both scenarios are supported - you can inspect gcode before sending as well as just slice+send right away. We are always considering various design choices and the Octoprint feature in particular has been made with community input. Of course, there are always ways to improve, but perhaps you might want to invest some consideration yourself in order to see how different design choices affect other users.

MartinMajewski, 2019-01-25:
@vojtechkral Okay, I get it! I don't want to mess with other's people requirements.

So my last proposition to also get people on board that have the same requirements on that function as I have:

Why not place a checkbox in the printer settings (or beside the "Send to printer" button or within the dialog itself) that enables/disables the slicing function? One that needs this operation can enable the "slice on send to printer host" or vice versa. I mean, this is not a big and deep change. It requires just to check wheater or not the slicing functionality is enabled and perform it or leave it.

bubnikv, 2019-01-28:
Smart substitutions will be disabled in the edit field in the Slic3r PE 1.42.0-alpha4 build.

bubnikv, 2019-02-01:
> Starting a slicing process is a hidden side effect (which you are not stating anywhere, nor can it be aborted easily, because slic3r's cancel feature is unreliable!)

Canceling should be reliable now. If it is not, then it is a bug and it should be fixed.

> OctoPrint locks the file and you have to manually abort the print job in OctoPrint - which is totally legit from OctoPrints side, but a big UX fail in slic3r PE.

I don't understand. The OctoPrint upload thread in Slic3r PE 1.42.0-alpha4 works with a copy of the generated G-code, it is only the copy that gets locked.

bubnikv, 2019-02-01:
> The problem is (as shown by the screenshots) that Slic3r does not put a .gcode file extension behind the filename.

It does.

> Also, it does not check for forbidden characters which are e.g. existing when the filename originates from Fusion 360 directly. 

Those characters form a valid file name, therefore they should be valid, though as you state, the file name generated by Fusion 360 seems ugly to you.

The current master allows you to rename the object.

> Moreover, the filename string is split into hidden parts when a hyphen is present.

This is a bug in wxWidgets, which was fixed today by patching wxWidgets 3.1.1 on the Prusa3d build server prusa3d/wxWidgets@a832410.

> The entire name gets selected, which leads to having the extension deleted easily with one press of a button.

Just the file name, not the extension gets selected. It should now work correctly with the patched wxWidgets.

Closing.

bubnikv, 2019-02-01:
See https://github.com/prusa3d/Slic3r/issues/1575 for the wxWidgets input line issues.

closed 2019-02-01 by bubnikv

MartinMajewski, 2019-02-01:
> I don't understand. The OctoPrint upload thread in Slic3r PE 1.42.0-alpha4 works with a copy of the generated G-code, it is only the copy that gets locked.

What I mean is that as soon as the G-code is uploaded to OctoPrint with the "Start printing after upload" option enabled OctoPrint will lock the G-code to perform the print job. When you realize an issue with the sliced object, you have no authority to alter the G-code in OctoPrint anymore. You have to switch to OctoPrint and cancel the job there to release the G-code file for overwrites.

I said this in the context of usually checking the sliced object before uploading.

Okay, use-cases may vary, this is fair enough. 
But I think having all options available and visible is definitely a UX plus.

As the MK3 is featuring an even tighter integration of an RPi and therefore OctoPrint, there should be a more elaborate integration of OctoPrint either way. Just my two cents...

bubnikv, 2019-02-01:
>  But I think having all options available and visible is definitely a UX
plus.

That is a matter taste. Often having all options available and visible is
an UX mess.

commit 81da613c (2019-02-18): Printhost: Persist upload path & start print checkbox (re-add lost code)
Fixes #1219
Fixes #1004
Fixes #1106
Fixes #1678
commit 08f1459a (2019-02-19): Printhost: Persist upload path & start print checkbox (re-add lost code)
Fixes #1219
Fixes #1004
Fixes #1106
Fixes #1678