prusa3d/PrusaSlicer #1680: [Request] Custom filename - used_filament in meters

DaTTcz, 2019-01-18:
The output file name parameter "used_filament" is in millimeters with three decimal places? Better would be in meters I think. Just like on the panel.

bubnikv, 2019-01-29:
Implemented with de742bbab2a81ee182fc707754227dcbdbfa5869

DaTTcz, 2019-02-09:
Thank you. Slic3rPE 1.42.0 a5 - Now it is stored in meters but does not round off. For 3.00 on panel is stored 2.99615. Better to round them to 2 decimal places.

bealparslan, 2019-05-31:
> Now it is stored in meters but does not round off. For 3.00 on panel is stored 2.99615. Better to round them to 2 decimal places.

+1 Same problem here. In a nut shell; **{used_filament}** stored in (ex.) 5.66696 meters. I prefer round up to 5.67
Like the sliced info section.

bealparslan, 2019-05-31:
+ **{total_weight}** needs rounding too.

sebaschn, 2020-03-05:
+1
just encountered the option to specify the output names. but the whole decimal value isnt human readable.
please fix :-)

jvanbalen, 2020-04-07:
Easy fix: use "..._{int(total_weight)}g_..."
For more decimals use "..._{int((total_weight*10.0))/(10.0)}g_..." or replace the "10.0" with any order of magnitude you like.
And yes, the amount of numbers is important.
Tested with Slicer 2.2.0

mathiasvr, 2020-12-02:
Thanks @rodeo-sam, I'm using this to display filament meter usage in the filename: `{int((used_filament*10))/10.0}m`.

bubnikv, 2020-12-03:
What we need is some kind of general number to string formatting function. That is not a 5 minutes task though, as our homebrew macro processor is fast and efficient, but not very easy to extend.

theslimshaney, 2021-01-06:
> Easy fix: use "..._{int(total_weight)}g_..."
> For more decimals use "..._{int((total_weight*10.0))/(10.0)}g_..." or replace the "10.0" with any order of magnitude you like.

@rodeo-sam this is truncating not rounding. If your weight was like 1.99, that would make it show as 1.9 instead of 2.
You can round like this `{(int((total_weight * 10.0 - 0.5)) + 1) / 10.0}`