prusa3d/PrusaSlicer #1686: Geometry print starts mid-air after support geometry was printed first completely (unintentional!).

MartinMajewski, 2019-01-20:
### Version
1.42.0-alpha3

### Operating system type + version
macOS 10.14.3

### Behavior
Please find attached the compressed configuration, gcode and STL files.

Out of a sudden, the sent to OctoPrint gcode produced an odd behavior: It firstly printed the entire support structure for the four screw holes and started the print of the actual geometry then. However, it started printing the geometry from the height position of the topmost support layer, thus, printing mid-air.

#### STL/Config (.ZIP) where problem occurs

[ElectronicsBox-RepRapDiscount-FGSC.zip](https://github.com/prusa3d/Slic3r/files/2776404/ElectronicsBox-RepRapDiscount-FGSC.zip)

bubnikv, 2019-01-21:
Do you see the issue in the G-code? Has it something to do with OctoPrint?
Please understand, that we cannot invest developer's time into investigating such vague error reports. If you help us with the investigation, the developer time will be invested into fixing other issues or implementing new features. Thank you.

bubnikv, 2019-04-13:
no response, closing

closed 2019-04-13 by bubnikv