prusa3d/PrusaSlicer #1687: Send to printer requires Restart 1.42.0 Alpha 3+

fwdixon, 2019-01-20:
### Version
1.42.0 Alpha 3+ win 64 version

### Operating system type + version
Windows 10 x64

### Behavior
If a printer is selected that does not have an octoprint server set up, you must select an octoprint-enabled printer and then restart slic3r for the "send to printer" button to be displayed.
[image]
[image]

fwdixon, 2019-01-20:
Update: Making any changes to settings causes the "send to printer" button to disappear (in my case, changing infill %)

bubnikv, 2019-01-29:
@YuSanka would you please look into this issue? Thanks

fwdixon, 2019-02-16:
Just downloaded the latest Alpha 5, this issue is still present.

vojtechkral, 2019-02-18:
Can someone reproduce the problem? I can't seem to. The moment I select an Octoprint-configured printer, the _Send_ button show up.

vojtechkral, 2019-03-04:
Finally managed to reproduce the problem. Will try to fix...

closed 2019-03-05 by vojtechkral

fwdixon, 2019-03-05:
Will test when I get back in to the office next week.

fwdixon, 2019-03-16:
Issue is fixed in Alpha 7. Thanks!

commit f7529dbe (2019-01-29): Tab: Update print host address and cafile properly, fix #1687
commit f84f916e (2019-03-05): PrintHost: Fix host type configuration, refactor, fix #1687
commit 857b68a8 (2019-03-05): PrintHost: Fix host type configuration, refactor, fix #1687