prusa3d/PrusaSlicer #1688: Feature request: Rotate separate parts after cloning

Armadillo1, 2019-01-20:
After duplicating parts and trying to rotate them (+-45 deg), all of them start turning instead of the one selected. Would be nice to allow separate rotation per part selected.

paynterf, 2019-01-23:
Same thing for 'Mirror'.  Mirroring a duplicated object mirrors all objects, not just the one selected

bubnikv, 2019-01-29:
It is now possible to detach instance or multiple instances into independent objects from a context menu in the 3D scene, from the instance context menu in the object panel, and by dragging the instance inside the object panel to an empty space.

https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-alpha4

Closing

closed 2019-01-29 by bubnikv