prusa3d/PrusaSlicer #1689: How to access single values of multi-dimensional arrays?

fishcu, 2019-01-20:
### Version
Latest alpha release, 1.42 a3

### Operating system type + version
Win 10

### Behavior
I'm trying to access the bed size Y value, so that I can present the print after it is finished. The desired end gcode would include this line:
`G1 X0 Y230 F1500 ; present print`

I can get halfway there by putting
`G1 X0 Y{bed_shape[2]} F1500 ; present print`
, however that will produce the following gcode:
`G1 X0 Y[230.000000, 230.000000] F1500 ; present print`

Now, I tried to access the second value only like this:
`bed_shape[2][1]`
and
`bed_shape[2, 1]`
, but neither of these indexing methods work.
Neither does something like
`bed_shape[2].y`
, which would have been my next guess.

Short of reverse-engineering the variable replacement parser, how can I get the bed_shape Y size?

fishcu, 2019-01-20:
I had a look at the code, and it seems like parsing of nested arrays is not implemented. 
If I understood correctly, the corresponding code starts at `PlaceholderParser.cpp:637`, including the `legacy_variable_expansion` and `legacy_variable_expansion2` functions.

Can anyone confirm my findings?

bubnikv, 2019-01-20:
it is not implemented.

bubnikv, 2019-01-21:
Actually there is already one.
https://github.com/prusa3d/Slic3r/issues/990
Closing this issue as a duplicate of https://github.com/prusa3d/Slic3r/issues/990

closed 2019-01-21 by bubnikv