prusa3d/PrusaSlicer #1690: Gyroid and 3D honeycomb infill not correctly connected to perimeters

wavexx, 2019-01-20:
### Version
4ffe76f012529185aa7834adea53087063b3959f

### Behavior
While investigating #1451 I noticed that these extra lines are *not* correctly connected to the perimeter. On a perimeter curve, `gyroid` and `3d honeycomb` will simply trace a chord between the perimeter intersections instead of following the perimeter like `cubic` correctly does.

A curvy object with cubic:
[image]

Same layer with gyroid:
[image]

This is much more evident when looking from the zenith:
[image]

In addition, `hilbert`, `octagram` and `archimedean chords` do not even attempt to perform perimeter connection.

photog0411, 2019-01-20:
So is cuboid presently the best infill option?

wavexx, 2019-01-21:
On Mon, Jan 21 2019, photog0411 wrote:
> So is cuboid presently the best infill option?

I don't think this is the best place to discuss this. This depends
heavily on the piece you want to print, why you print it (maybe you
aesthetics matter) and how stress is applied. Using grid and aligning
the pattern can do better than cubic and be faster to print in some
cases.

wavexx, 2019-01-21:
@supermerill : Did you also write the 2d honeycomb generator? I created a PR to fix #1451 based on your work, and managed to fix also the "lines" infill. But `honeycomb` needs some extra tweaks in order to do something meaningful with that option

supermerill, 2019-01-21:
nop

supermerill, 2019-01-24:
done (commits for cherry-picking: supermerill@162132eb351e65a134cb0489ee89e8ce88a50d3a & supermerill@b07d2461d38c7112c56332724b9ceb87e456211d-

It's a bit slow with big geometry with many point, but i can't see how to improve that without compromising on quality.
This will be in the next version of my fork.

@bubnikv Do you want a pr for this?

wavexx, 2019-01-27:
@supermerill : any chance you can also fix the 3d-honeycomb wall attachment?

supermerill, 2019-01-27:
It's not?

wavexx, 2019-01-27:
No, it has the same problem as shown for gyroid. I checked all filling patterns.

supermerill, 2019-01-28:
can't reproduce, i spend a day to make it works.

wavexx, 2019-01-28:
Ok - I didn't realize they were connected.

mperino, 2019-02-01:
+1 for getting this fix into a mainline release.

wavexx, 2019-03-06:
@supermerill: I found some cases where the infill connection breaks.

[gyroid_test.zip](https://github.com/prusa3d/Slic3r/files/2938657/gyroid_test.zip)

Here the infill drops outside of the volume:

[image]

I also noticed that it often fails to generate the connection entirely in some areas:

[image]

supermerill, 2019-03-08:
corrected with supermerill@7146471c288f147eced5988219a6fc1b1a45f528

thanks for the bug report.

wavexx, 2019-03-08:
Thanks to you for looking into it.

Looking at the same model/config, there are still some missing connections. Look at layer 26/27, on top of the bottom/left hole:

[image]

there is a missing connection which is skipped for some reason. But if you go to layer 58 it pops up as it should.

wavexx, 2019-03-17:
@supermerill I made another minimal test-case based on the one above where there's an entire wall without perimeter connection:

[image]

Here's the minimal test case:

[test.zip](https://github.com/prusa3d/Slic3r/files/2975418/test.zip)

this one seems to happen at several layer heights and infill values as well.

supermerill, 2019-03-17:
thanks, i will see

wavexx, 2019-05-31:
I've found plenty of cases where the lack of infill connection makes the piece overall weaker than cubic: the piece will support a greater load, but walls will tear apart almost instantly with minimal shear.
This is especially visible with PLA.

I didn't follow your fork recently, did you had a chance to look at it?

kramer, 2020-03-24:
So this is now resolved with the public release of 2.2.0?

bubnikv, 2020-03-25:
I believe it is resolved in PrusaSlicer 2.2.0. Thanks @supermerill and @wavexx for initial work on this topic.

closed 2020-03-25 by bubnikv