prusa3d/PrusaSlicer #1691: Incorrect filament profile being used during purge - always set to PLA.

skellatore, 2019-01-21:
### Version
1.41.2+win64

### Operating system type + version
Windows 10, 64 bit

### Behavior
With a MMU Print, after slicing and exporting GCODE the wipe tower is using speeds/settings for "; material : DEFAULT (PLA)" as shown in the GCODE. In the example below, All my filaments and types were set to PET.

```
G1 Z20.000
G1 E1.20000 F3000.00000
;--------------------
; CP TOOLCHANGE START
; toolchange #10
; material : DEFAULT (PLA)
;--------------------
M220 S100
M907 E750
; CP TOOLCHANGE UNLOAD
G1 X35.175 Y111.355 
G1 E-15.0000 F6000
```

* _Steps needed to reproduce the problem_
Open a model in Slic3r using MMU2 printer profile and MMU2 Printer settings
Slice and export gcode
Review GCODE.

bubnikv, 2019-01-21:
Please attach your 3MF.

skellatore, 2019-01-22:
[Pokemon Logo Blue_0.25mm_PET_MK3.zip](https://github.com/prusa3d/Slic3r/files/2780902/Pokemon.Logo.Blue_0.25mm_PET_MK3.zip)
as requested.