prusa3d/PrusaSlicer #1693: [Feature Request

joeynuggetz, 2019-01-21:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

joeynuggetz, 2019-01-21:
Would be nice to be able to define custom purge volumes between colors and have them saved somewhere instead of having to set them for every print job. I like the advanced screen in the purged volumes options but would be nice if I could set this on each filament profile so I can just set colors and print.

skellatore, 2019-01-31:
@bubnikv https://github.com/prusa3d/Slic3r/issues/1282

rtyr, 2021-01-26:
Duplicate of #1282. Closing.

closed 2021-01-26 by rtyr