prusa3d/PrusaSlicer #1694: Error compiling on Linux with -DSLIC3R_WX_STABLE=1 (WX Widgets 3.0)

sebastianha, 2019-01-21:
### Version
version_1.42.0-alpha3-71-ga7c876a9b

### Operating system type + version
openSUSE-Leap-15.0

### Behavior
When compiling with "-DSLIC3R_WX_STABLE=1" I get the following error:

`Slic3r_prusa/src/slic3r/GUI/OptionsGroup.cpp:237:9: error: ‘wxGETTEXT_IN_CONTEXT’ was not declared in this scope
         wxGETTEXT_IN_CONTEXT("Layers", wxString(option.label)) :`

Is this break intentional? This method has been introduced in 3.1. It has been introduced in commit 1a474ea2820776e1a1510fc494ed8508a59d6944

stelgenhof, 2019-01-22:
@sebastianha I have the same issue. (Running Arch Linux).

 I haven't installed wxWidgets 3.1 yet since that package for Arch Linux has not been provided. For it to work, I would need to install the development branch of wxWidgets, which I am a bit hesitant about.

vojtechkral, 2019-01-22:
Yes, I thought this would happen but didn't fix yet. I'll have a look.

closed 2019-01-23 by vojtechkral

stelgenhof, 2019-01-23:
Yes, compiles perfectly now. Thanks @vojtechkral!

commit fde299bf (2019-01-23): I18N: wx 3.0 back-compat for translations with context, fix #1694