prusa3d/PrusaSlicer #1698: Linux: Add support for using symlinks to the slic3r script

Jyx, 2019-01-22:
I've noticed that you cannot symlink to the `slic3r` script in the pre-built package, since the script will refer to the location of the symlink instead of the folder where Slic3r PE actually is. I've cloned the project/source with intention to make a fix and send a pull request. But after having a look I cannot figure out where the `<root>slic3r` script it put together (burried down in the Makfiles?). If someone here could guide me or eventually just take this patch and apply it, then people can start to use symlinks to slic3rPE on Linux.

```patch
--- Slic3rPE-1.42.0-alpha3+linux64-full-201901111425/slic3r	2019-01-11 15:44:51.000000000 +0100
+++ 1.42.0-alpha3/slic3r	2019-01-22 10:55:56.111370383 +0100
@@ -1,4 +1,4 @@
 #!/bin/bash
-DIR=$(dirname "$0")
+DIR=$(readlink -f "$0" | xargs dirname)
 export LD_LIBRARY_PATH="$DIR/bin"
 exec "$DIR/bin/slic3r" "$@"
```

bubnikv, 2019-01-29:
thanks, I have updated the build server.

closed 2019-01-29 by bubnikv

Ghostbird, 2019-02-17:
Was this a regression? This feature is already present in my 1.41 install, but it is referenced as a bugfix on the 1.42 alpha release summary.