prusa3d/PrusaSlicer #1699: Possible wrong version displayed after compiling 1.42.0-alpha3 (shows up as alpha2)

sgano, 2019-01-22:
### Version
_Version of Slic3r used goes here_ 1.42.0 alpha 3 (--branch version_1.42.0-alpha3)

### Operating system type + version
MacOS  10.14.2

### Behavior
To enable dark mode on MacOS 10.14 (Mojave) I have compiled Slic3r PE manually.  I cloned the repository using: git clone --branch version_1.42.0-alpha3 git://github.com/prusa3d/Slic3r

However, after successfully compiling it - the application says it is version "1.42.0-alpha2+UNKNOWN".  Did I pull from the wrong tag/branch?   Or is there a typo in the source?

This isn't a big deal and it is probably user error; but it was unexpected.  

Thanks for all the hard work!  Other than that the compilation went very smoothly on my Mac and the dark mode looks terrific!

_Is this a new feature request? No

sgano, 2019-01-22:
I have also tried these clone commands and still get the version name of alpha 2.
git clone --branch version_1.42.0-alpha3 --depth 1 git://github.com/prusa3d/Slic3r
git clone --branch version_1.42.0-alpha3 --single-branch git://github.com/prusa3d/Slic3r

supermerill, 2019-01-23:
yeah, the commit for the string change hasn't been pushed to the repo.
Not that difficult to change, just replace alpha2 by alpha3 in all files.

sgano, 2019-01-23:
Okay thanks!  I was starting to wonder if I had compiled the correct version so that helps explain the difference.

closed 2019-01-23 by sgano