prusa3d/PrusaSlicer #1700: 1.42.0-alpha 3 crash

thomaspj1017, 2019-01-23:
### Version
1.42.0-alpha3+win64

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 7

### Behavior
* _Describe the problem_
Slic3r crashes when trying to view the sliced file in the tool colors on the preview window.
* _Steps needed to reproduce the problem_
when I slice multimaterial files then try to view them by tool after being sliced it crashes. I have tried various dual color model and I get the same result.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
View the sliced model with different colors for the tool used there
* _Actual Results_
Slic3r crashes
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

FidelCapo, 2019-01-23:
Thank you for report. Unfortunately, I am not able to reproduce this issue. Can you please send me a 3mf file with the object causing this issue? Thank you

bubnikv, 2019-03-14:
No response, closing

closed 2019-03-14 by bubnikv