prusa3d/PrusaSlicer #1701: Layers editing

espr14, 2019-01-23:
### Version
1.42.0-alpha3+win64
201901111426

### Operating system type + version
Win 7 64

### Behavior
* Changing layers height for specific z regions works but when you move pointer out of layer height bar graph (lost of focus) it resets. It also resets after few seconds when not holding any mouse key.
* Reproduce: Open any 3D model, click Layer editing icon, change height of some layers and wait or move your mouse out.
* Expected: Changes will be kept
* Actual: Changes are reset

CageFox, 2019-01-27:
Same thing, unable to edit layer heights

Garfield1970, 2019-01-28:
Confirmed @ version 1.42.0-alpha3+win64 running on Windows 10 x64

bubnikv, 2019-01-28:
Fixed in 1.42.0-alpha4

closed 2019-01-28 by bubnikv