prusa3d/PrusaSlicer #1702: Extruder selection for bodies? I needed to use one filament for two of four bodies, could not find a way to set it.

TNDavid, 2019-01-23:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2019-01-29:
You mean instances?

Instances share all parameters. If you want the parameters to differ, you have to split the instances, this is now supported in Slic3r 1.42.0-alpha4

TNDavid, 2019-01-31:
When I exported the separate .stl files of a multi-body part in SolidWorks they are called bodies.  In Slic3r 1.42.0 Alpha 3 I could find no way to do it.  Someone on another group told me that the filament number could be changed in expert mode by double clicking on it. It worked, but very difficult to get the number selection to highlight when clicking, had to highlight the line, and try multiple times to get it to work. It is very tricky. That hopefully will get fixed.

closed 2019-01-31 by TNDavid