prusa3d/PrusaSlicer #1703: SIGSEGV due to Gtk file chooser on ArchLinux

balert, 2019-01-23:
### Version
version_1.42.0-alpha3-91-g5a707a66b

### Operating system type + version
ArchLinux x64

### Behavior

SIGSEGV when opening file chooser dialog. 

gtk_file_chooser_dialog_new (libgtk-x11-2.0.so.0) appears in coredump. I believe the Slic3r is not compatible with my (ArchLinux) Gtk library?!

Slic3r crashes when opening .stl file, but drag and drop into Slic3r works fine. I can use "Send to printer", so Slic3r is basically still "usable" for me by avoiding file chooser dialogs :-)

vojtechkral, 2019-02-22:
Hi, does this still happen with a more recent alpha?
Can you provide a backtrace?
Thanks!

balert, 2019-02-25:
yes, I've just tried tags/version_1.42.0-alpha6. Does this help:

Stack trace of thread 15358:
#0  0x0000001000000010 n/a (n/a)
#1  0x0000557696adf89f n/a (/home/user/path/Slic3r/build/src/slic3r-gui)
#2  0x0000557696af90e1 n/a (/home/user/path/Slic3r/build/src/slic3r-gui)
#3  0x0000557696ae0092 n/a (/home/user/path/Slic3r/build/src/slic3r-gui)
#4  0x00007f8d4227967b n/a (libpixbufloader-png.so)
#5  0x00007f8d42279d22 n/a (libpixbufloader-png.so)
#6  0x00007f8d437548c5 n/a (libpng16.so.16)
#7  0x00007f8d437558ac png_process_data (libpng16.so.16)
#8  0x00007f8d422783fa n/a (libpixbufloader-png.so)
#9  0x00007f8d456a0c45 n/a (libgdk_pixbuf-2.0.so.0)
#10 0x00007f8d456a0e35 gdk_pixbuf_loader_close (libgdk_pixbuf-2.0.so.0)
#11 0x00007f8d456a189c n/a (libgdk_pixbuf-2.0.so.0)
#12 0x00007f8d456a18fd gdk_pixbuf_new_from_stream (libgdk_pixbuf-2.0.so.0)
#13 0x00007f8d45bd0a62 n/a (libgtk-x11-2.0.so.0)
#14 0x00007f8d45bd3d5a gtk_icon_info_load_icon (libgtk-x11-2.0.so.0)
#15 0x00007f8d45bd3ee6 gtk_icon_theme_load_icon (libgtk-x11-2.0.so.0)
#16 0x00007f8d45bcf427 gtk_icon_set_render_icon (libgtk-x11-2.0.so.0)
#17 0x00007f8d45d23ef1 gtk_widget_render_icon (libgtk-x11-2.0.so.0)
#18 0x00007f8d45bb33af n/a (libgtk-x11-2.0.so.0)
#19 0x00007f8d450f906b n/a (libgobject-2.0.so.0)
#20 0x00007f8d450fa75d g_object_new_with_properties (libgobject-2.0.so.0)
#21 0x00007f8d450fa86a g_object_new (libgobject-2.0.so.0)
#22 0x00007f8d45bb7cd5 n/a (libgtk-x11-2.0.so.0)
#23 0x00007f8d450f906b n/a (libgobject-2.0.so.0)
#24 0x00007f8d450fa75d g_object_new_with_properties (libgobject-2.0.so.0)
#25 0x00007f8d450fa86a g_object_new (libgobject-2.0.so.0)
#26 0x00007f8d45bb4955 n/a (libgtk-x11-2.0.so.0)
#27 0x00007f8d450f906b n/a (libgobject-2.0.so.0)
#28 0x00007f8d450f9ea4 g_object_new_valist (libgobject-2.0.so.0)
#29 0x00007f8d450fa83a g_object_new (libgobject-2.0.so.0)
#30 0x00007f8d45bb4dbe n/a (libgtk-x11-2.0.so.0)
#31 0x00007f8d45bb4eea gtk_file_chooser_dialog_new (libgtk-x11-2.0.so.0)
#32 0x00005576964942d0 n/a (/home/user/path/Slic3r/build/src/slic3r-gui)
#33 0x00005576964957fb n/a (/home/user/path/Slic3r/build/src/slic3r-gui)
#34 0x0000557696214aa5 n/a (/home/user/path/Slic3r/build/src/slic3r-gui)
#35 0x000055769623f95e n/a (/home/user/path/Slic3r/build/src/slic3r-gui)

vojtechkral, 2019-02-28:
@balert Thanks. I can't say for sure but it seems like memory allocation callbacks are botched for gtk in libpng. I'll see what can be done...

bubnikv, 2019-03-02:
What happens if you download the .bz2 archive instead of the appimage, you remove the libpng from the unpacked directory and then you run Slic3r?

It may be, that the bundled libpng12 fights with the one installed on your system.

balert, 2019-03-04:
yes, it works from the bz2 archive. that's my current work around.

vojtechkral, 2019-03-28:
@balert We've removed a direct libpng dependency. Would you please re-test with the following AppImage?

[Slic3rPE-1.42.0-beta1+78-linux64-full-gd209aaa-201903281040.tar.gz](https://github.com/prusa3d/Slic3r/files/3017594/Slic3rPE-1.42.0-beta1%2B78-linux64-full-gd209aaa-201903281040.tar.gz)

balert, 2019-03-30:
@vojtechkral the AppImage did apparently work well. Had no crash due to file chooser anymore. Great work.  Thanks!

bubnikv, 2019-04-01:
Great. So the crash was likely due to the libpng bundled with Slic3r not to be (sic!) binary compatible with the libpng of the same version installed on your system.

We now link the libpng statically only, so this issue is gone. Thanks for testing.

closed 2019-04-01 by bubnikv