prusa3d/PrusaSlicer #1704: File generation error

landrunner22002, 2019-01-24:
### Version
_Version of Slic3r used : 1.41.2

### Operating system type + version
Mac OS Sierra 

### Behavior
After selecting "Polymaker PC-MAX" in "Filament" menu, the exported Gcode file from Slic3r 1.41.2 added "ABS" as filament type instead of "PC-MAX" or "PC" in file name.

bubnikv, 2019-01-24:
because it is using the key "Filament type" from the selected filament,
which is set to ABS.

landrunner22002, 2019-01-24:
> 
> because it is using the key "Filament type" from the selected filament, which is set to ABS.
> […](#)

And no PC-MAX is available in "Filament Type" menu

bubnikv, 2019-01-26:
No. The filament type should be a generic identifier, for PC-MAX a PC would be appropriate.

Originally this field was added to support the Prusa Multi Material upgrade, so that the wipe tower generator would decide the ramming sequences on it. Now we have the ramming sequences parametrized, but there is still some functionality decided internally based on the filament type, namely the print slows down when mixing flex or soluble material with non-flex.

You are right, we shall extend the list of generic polymers, or we should make the list extensible / allow free form texts here.

jhoblitt, 2019-01-27:
It is also useful for the filament type to be encoded in the in the gcode filename.

jhoblitt, 2019-01-27:
Off the top of my head, here are a few other polymers in use:

- nylon or PA
- polypropylene or PP
- PEI
- PEEK
- PEKK (different from PEEK)
- POM
- PSU
- PVDF

jhoblitt, 2019-01-27:
@bubnikv Do additional material types only need to be added to `src/libslic3r/PrintConfig.cpp` or is there additional work required?

jhoblitt, 2019-01-27:
oh, hmm, looks like the ConfigOptionEnum template also needs to be updated.

jhoblitt, 2019-01-27:
uh, are two enums for the filament necessary?

supermerill, 2019-01-29:
I also had a request for "htpla", i think a free-form field text field (or an extensible list) is needed.

photog0411, 2019-02-02:
> 
> 
> I also had a request for "htpla", i think a free-form field text field (or an extensible list) is needed.

+1

photog0411, 2019-02-02:
> 
> 
> It is also useful for the filament type to be encoded in the in the gcode filename.

precisely... it's very useful for those of us that do frequent prints with different materials

akust, 2019-05-13:
also for me as Palette2 user, I'm afraind I need a string to define distinct "from XFilament to YFilament" setting (pre slicing settings) in the gcode (see p2pp project). Concretely I'm stuck now with BVOH what runs under the type "PLA" in Slic3r but definitly needs a special slicing setting. It seems Slic3r enables me to enter my own string (e.g. "BVOH") but after some other changes it turns out it ignored my input and the string is back to "PLA"..

bubnikv, 2019-05-17:
Implemented with https://github.com/prusa3d/PrusaSlicer/commit/c7f6e5436c0215d34ed4ba417288876fcde02df5
Also the freeform editing of the field is now supported with https://github.com/prusa3d/PrusaSlicer/commit/f0acf4f745e65798fc4bd82acaf40bb756937cda
see https://github.com/prusa3d/PrusaSlicer/issues/2232
Closing

closed 2019-05-17 by bubnikv

bubnikv, 2019-05-19:
@jhoblitt thanks for the list, I have added it verbatim.
I sorted them so that the first 5 most commonly used are at the top and the rest is sorted alphabetically.