prusa3d/PrusaSlicer #1705: Česká verze hustota filamentu

hartpa, 2019-01-24:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2019-01-24:
co je špatně?

hartpa, 2019-01-24:
Při zadávání hustoty v uživatelském nastavení: Při zadávání nelze použít čárku jako oddělovač desetinných míst (u jiných údajů lze)

hartpa, 2019-01-26:
Nastavení filamentu -> Filament -> Hustota: zadám hodnotu 0,76 (oddělovač desetinných míst čárka) a pak uložím uživatelské nastavení. Pak načtu jiné nastavení. Když znova načtu své původní uživatelské nastavení hustota 0,76 se neobjeví. Když udělám totéž, ale pro oddělení desetinných míst použiji tečku chová se program jak bych očekával (=změněná hodnota se uloží).

bubnikv, 2019-01-29:
Implemented with c973500ee65c92dc99b7a8177c75604f2966bc4c
Numeric input fields will support both dot and comma as a valid decimal separator.

closed 2019-01-29 by bubnikv